ArduPilot/ardupilot #13462: Quadcopter doesn't take-off

Ashlee-11, 2020-02-03:
#### Issue details
I recently assembled together Robokits DIY drone kit. Pixhawk board is of version 2.4.8 and battery is 2200Mah Li-Po battery. I uploaded arducopter firmware version 3.4.6 taken from this site https://firmware.ardupilot.org/Copter/stable-3.4.6/PX4-quad/ [Arducopter-v2.px4]. All calibrations were then done successfully. When I tried to to fly it though, it refuses to leave the ground even at 100% throttle.
The only parameter I changed was FS_Batt one from 10.5V to 6V as my battery was charged to 9.4V  even after 10hrs so I thought my battery's upper limit is 9.4V. Is the issue with battery or something else? It's a brand new battery, if it's the issue with battery need suggestions on which battery to look out for in the future.   

#### Version
Firmware version 3.4.6
Cinfiguration done with the help of Mission Planner version 1.3.70

#### Platform
Quadcopter

#### Airframe type
Quad X

#### Hardware type
PX4FMU V2

khancyr, 2020-02-03:
Hello,

Please open an issue on https://discuss.ardupilot.org/ for support questions. We reserve this place for code and improvement only.
About your question, yes that is probably an issue with your battery or esc. Please check the voltage rating of you motor and if you need to calibratr your esc.

closed 2020-02-03 by khancyr

rmackay9, 2020-02-03:
@Ashlee-11,

If it's got a Pixhawk 2.4.8 on there then I think you should try and put Copter-4.0 on there and then you can report any issues in the Copter-4.0 support area: https://discuss.ardupilot.org/c/arducopter/copter-40/155