ArduPilot/ardupilot #13464: Copter: implement Circle within Guided (aka LOITER_TURNS)

rmackay9, 2020-02-03:
This enhancement request comes from this discussion: https://discuss.ardupilot.org/t/copter-not-supporting-mav-cmd-nav-loiter-turns-question/51785

The request is to implement Circle mode (aka LOITER_TURNS) from within Guided mode.

Support for LOITER_TURNS is already implemented in Auto mode so this should be possible.

arshPratap, 2021-02-09:
I have a PR(#16473) that works on the issue. Would be nice, I could get some suggestions on it . Thanks in advance!!

shiv-tyagi, 2022-01-07:
I decided to work on this since I see no activity here for almost a year. Thanks @arshPratap for all you work :-)

arshPratap, 2022-01-07:
Thats great to hear !
I am sorry I wasnt able to work on some of my PRs as I am currently busy with some other commitments.
@shiv-tyagi  A helpful advice would be to look and understand some of the issues that were present in my PR (specially the part where the CIRLE code was reused from a prior segment and the suggestion was to generalize that part of code)
Looking forward to your PR!
Thanks

shiv-tyagi, 2022-01-07:
@arshPratap Thanks! I saw the issues with your PR and tried to address them in my PR #19718. Please have a look.

shiv-tyagi, 2022-01-10:
@rmackay9 Can you please review my PR? Thanks in advance.

adityaomar3, 2023-09-09:
@shiv-tyagi is this issue still unresolved ? I would like to work on it

shiv-tyagi, 2023-09-09:
@adityaomar3 I think this is still open. I actually submitted #19718 was intended to close this but I couldn't get it in. Feel free to refer to that branch. All the best.