ArduPilot/ardupilot #13471: Copter - Zigzag mode arm (fails), then flick to loiter instantly arms aircraft

JacksonUAS, 2020-02-03:
### Bug report

Arming the aircraft with CSC in zigzag mode - does nothing
and then flicking to loiter mode copter instantly arms. 

This has the potential to be extremely dangerous. 

**Version**
4.0.1

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Octo

**Hardware type**
Cube black

rmackay9, 2020-02-03:
@JacksonUAS, thanks for the report.

ZigZag mode is not an armable mode but I think it could be made to be fairly easily (we just need to make sure it handles all the PILOT_TKOFF_ALT, etc like other modes).

Were you using the transmitter sticks or auxiliary switch to arm the vehicle?

JacksonUAS, 2020-02-03:
Yeah, I was standing next to the pilot, and explaining how Zigzag is not an armable flight mode, and instructed him to try it (via combined stick commands) and see, and obviously the aircraft didnt arm. He then changed modes to loiter, and the aircraft instantly armed without any input other than the mode change.

peterbarker, 2020-02-04:
@jackcarrozzo Unable to trivially reproduce.

Can you provide some logs from when that happened, please?

peterbarker, 2020-02-04:
I might say that if you did continue to have the rudder over in the "arm vehicle" position when he flicked to loiter that it could arm with only a very small delay.

JacksonUAS, 2020-02-04:
> @jackcarrozzo Unable to trivially reproduce.
> Can you provide some logs from when that happened, please?

Will be here somewhere, sorry Im still having MP issues with this log. Keeps crashing on opening it, Im about to do a fresh install. 

https://drive.google.com/open?id=1c-LLlP57p3mZc9YSZi9k-11x-F5jAZAc

> I might say that if you did continue to have the rudder over in the "arm vehicle" position when he > flicked to loiter that it could arm with only a very small delay.

I wasnt holding the transmitter, but I dont think so. Was trying to make an example of how it wouldnt arm in ZigZag mode, was a few seconds delay post trying to arm to flicking it into loiter and having it leap into life.

peterbarker, 2020-02-04:
At first glance I can't see the "immediate arm" problem from the first graphs I made:

[image]

An arm event has happened when the red line is at its lowest point; disarm just above that.

None of that happens the instant loiter is entered.

The second zig-zag block there appears to have full-right-rudder for a couple of seconds - but it seems to have been flying when that happened....

Can you remember when this happened?

Also - were you landing in the same spot each time?

[image]

seems to show each time you took off it was from a slightly higher elevation.

JacksonUAS, 2020-02-06:
Think Ive found it - Aircraft never actually disarmed, despite pilot going through the motions of disarming. Was landed in ZigZag, and the props were stopped on the aircraft, flicking back into loiter instantly started them again as it was never actually disarmed??!

[image]

JacksonUAS, 2020-02-06:
> The second zig-zag block there appears to have full-right-rudder for a couple of seconds - but it seems to have been flying when that happened....
> 
> Can you remember when this happened?
> 
> Also - were you landing in the same spot each time?

Nope, were taking off and landing at various spots around park trying to figure out lurching issue.

> seems to show each time you took off it was from a slightly higher elevation.

- Baro drift I guess?

Looks like the problem isnt actually it arming, its that it appears like it is disarmed when landed in ZigZag, despite not being disarmed.

rmackay9, 2020-02-06:
I think the issue may be because MOT_SPIN_ARM = 0 (the default is 0.1) meaning that the motors won't spin when the vehicle is armed.

JacksonUAS, 2020-02-06:
Curious... If throttle was at 1500uS, they should be spinning in ZigZag while on the ground and armed though?

rmackay9, 2020-02-06:
They should but they don't apparently and the reason is that ZigZag has a simplified version of the spool logic and it's [missing a set-desired-spool-mode here](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_zigzag.cpp#L42).

We can reproduce the issue in SITL by doing this:

- param set MOT_SPIN_ARM 0
- graph SERVO_OUTPUT_RAW.servo1_raw (to graph the 1st motor's outputs)
- loiter (to switch to loiter mode)
- rc 3 1000 (to reduce throttle to zero)
- throttle arm (to arm the vehicle) - the motors don't spin
- rc 3 1500 - the motors spin-up to about 1150 - the MOT_SPIN_MIN value
- zigzag (to switch to zigzag mode) - the motors stop spinning
- loiter - the motors spin-up to 1150

This is an easy fix but I suspect users will still be annoyed they can't takeoff in zigzag mode no matter how high they raise the throttle so I think we need to do this larger fix.

rmackay9, 2020-02-06:
@JacksonUAS,

I've got a PR to add arming, takeoff and landing to ZigZag mode which also fixes the bug you've uncovered.  Once it passes peer review, if you'd like I could create a binary that includes this change on top of Copter-4.0.2-rc4.  That would help increase confidence that the change is good and we could include it in the next point release (aka Copter-4.0.3)

JacksonUAS, 2020-02-06:
> @JacksonUAS,
>
> I've got a [PR to add arming, takeoff and landing to ZigZag mode](https://github.com/ArduPilot/ardupilot/pull/13501) which also fixes the bug you've uncovered. Once it passes peer [...]

Awesome, happy to test it.

peterbarker, 2020-02-06:
On Wed, 5 Feb 2020, JacksonUAS wrote:

> Think Ive found it - Aircraft never actually disarmed, despite pilot going through the motions of disarming. Was landed in ZigZag, and the props were
> stopped on the aircraft, flicking back into loiter instantly started them again as it was never actually disarmed??!

I did actually consider that scenario.  I checked your MOT_SPIN_MIN 
parameter was reasonable - 15%.

Is 15% insufficient for your props to turn?  Do they spin when you first 
arm the aircraft at the "min" level?

rmackay9, 2020-02-06:
@peterbarker, it's the MOT_SPIN_ARM that is the key parameter..

rmackay9, 2020-02-20:
I'm going to close this now because the PR to add arming, takeoff and landing to ZigZag has gone into master and is also included in Copter-4.0.3-rc1. https://discuss.ardupilot.org/t/copte-4-0-3-rc1-available-for-beta-testing-critical-but-rare-fix-included/52583

closed 2020-02-20 by rmackay9

rmackay9, 2020-02-22:
@JacksonUAS, you may have already seen the release notice but in any case, if you get a chance to test [Copter-4.0.3-rc1](https://discuss.ardupilot.org/t/copter-4-0-3-rc1-available-for-beta-testing-critical-but-rare-fix-included/52583) that'd be great.

JacksonUAS, 2020-02-22:
@rmackay9 - Yup have scheduled in with the octocopters owner to test it on Monday