ArduPilot/ardupilot #13472: Copter: Zigzag mode lurching in flight, loiters fine

JacksonUAS, 2020-02-03:
#13282 # Bug report

**Issue details**

Flying in Zigzag mode, either going from point A to B, or in "manual" zigzag flight is resulting in the aircraft aggressively lurching in flight. 

Aircraft behaves fine in loiter mode. 

Note - local copy of mission planner is crashing when trying to review log, will update if I can find anything once able. 

**Version**
AC 4.0.1

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
octa

**Hardware type**
Cube Black

**Logs**

https://drive.google.com/file/d/1c-LLlP57p3mZc9YSZi9k-11x-F5jAZAc/view?usp=sharing

JacksonUAS, 2020-02-06:
Looking at log for this, it appears the lurching coincides with gps primary changing?

rmackay9, 2020-02-06:
@JacksonUAS,

I think you're right that it's because of the GPS switch while the vehicle is moving between points A and B.  The graph below (taken from your log) shows the lurch.
[image]

This graph shows the sudden jump in the target velocity after the switch.
[image]

I suspect that our EKF is not elegantly handling the change in GPS.  It handles it much like it would handle a single GPS suddenly providing a jump in position.  If it's small then it will just absorb it, if it's large then it will wait a while for it to return to it's old location but if it doesn't then it jumps (and properly warns our navigation controllers of the jump).  In this case the jump is relatively small so it causes a lurch.

I've created an [issue to improve EKF's handling of the primary GPS changing](https://github.com/ArduPilot/ardupilot/issues/13503) but this will likely take time.  I think it should help to enable [GPS blending](https://ardupilot.org/copter/docs/common-gps-blending.html).

rmackay9, 2020-02-14:
I think I'll close this issue if that's ok because we've got the other one re [improving EKF's handling of GPS changes](https://github.com/ArduPilot/ardupilot/issues/13503).  Happy to re-open this one though if people object.

closed 2020-02-14 by rmackay9