ArduPilot/ardupilot #13478: AP_Servo: allow do-set-servo commands to affect more SERVOx_FUNCTION

rmackay9, 2020-02-05:
We should relax the check added in this PR https://github.com/ArduPilot/ardupilot/pull/9107 so that users can use do-set-servo to affect the outputs for those outputs where this is safe.

One user has requested do-set-servo commands be effective for the sprayer ([discussion](https://discuss.ardupilot.org/t/do-set-servo-in-spray-function-does-not-work-in-4-0-0/51950))

WickedShell, 2020-02-05:
This was actually intentional, though we actually talked about this kinda of case on the call. If we need a control mechanism for controlling the sprayer there should be explict commands for that. The servo relay events is weird because it easily crashes the vehicle, and is orthogonal to a libraries ability to actually control itself. This actually leads to weird problems where you toggle it, then the sprayer library may choose to move over it.

I'd personally prefer to see us not revert on this, and provide a more controlled/correct interface that the libraries are aware of when they need it.

rmackay9, 2020-02-05:
@WickedShell, I generally agree but because the extra effort wasn't put in to provide an interface at the time the restriction was added users see a functional regression.

rmackay9, 2020-06-11:
I'm going to close this because we've at least added the sprayer and gripper to the whitelist of functions that can be affected by do-set-servo

closed 2020-06-11 by rmackay9