ArduPilot/ardupilot #13483: WAF: command line switch --default-parameters does not work

Pedals2Paddles, 2020-02-05:
### Bug report

**Issue details**
If you invoke the command line argument `--default-parameters='file_path.param'` when running waf, it has no effect on the binary.  The specified parameter file not loaded, and the defaults.parm file in the hwdef is loaded instead.

I can't actually find where these command line switches are executed to fix it myself.  I suspect this is incredibly easy fix, just not sure where it is.

**Version**
Master

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

Pedals2Paddles, 2020-02-05:
Here is an example param file I try to use as defaults file on a CubeOrange build. 

[Hover1 Pre-Tune Param Diff.zip](https://github.com/ArduPilot/ardupilot/files/4157200/Hover1.Pre-Tune.Param.Diff.zip)

peterbarker, 2020-02-06:
Working for me.

Command-line was:
```
pbarker@bluebottle:~/rc/ardupilot(master)$ ./waf configure --default-parameters=xyzzy --board=Mini-Pix
pbarker@bluebottle:~/rc/ardupilot(master)$ ./waf build --target=bin/arducopter --upload
```

```
STABILIZE> param show LOG_BACKEND_TYPE
STABILIZE> LOG_BACKEND_TYPE 17.000000

pbarker@bluebottle:~/rc/ardupilot(master)$ cat xyzzy 
LOG_BACKEND_TYPE 17
```

Probably the commas in your source file; try replacing with a space each and retrying.

PR would probably be accepted to accept a comma as a delimiter rather than just whitespace.

Reopen if there is actually an issue :-)

Pedals2Paddles, 2020-02-07:
This is still a problem.  Two problems actually I think.

- If you specify something --default-parameters, it uses the git repo as the root.  So if I put a path in to a parameter file on my desktop, it tries to load this: `/home/matt/github/ardupilot/home/matt/Desktop/Hover1.param`. 

- If there is a defaults.param file in the hwdef directory, then the command line specified parameter file is completely ignored.

I'm happy to fix this. But someone needs to point me to where this code actually is?  It isn't in any directory in the ardupilot repo that I can find.

peterbarker, 2020-02-07:
https://github.com/ardupilot/ardupilot/blob/master/Tools/ardupilotwaf/chibios.py#L77

Pedals2Paddles, 2020-02-07:
This should take care of it. https://github.com/ArduPilot/ardupilot/pull/13512

closed 2020-02-13 by Pedals2Paddles