ArduPilot/ardupilot #13488: RTL does not descend to clear vertical fence breach

RhinoAus, 2020-02-05:
#### Issue details
Line 541 of mode_rtl.cpp prevents the vehicle from descending below the altitude fence if RTL is triggered as the fence breach behaviour. This is clearly intended functionality from lines 522 through 538

Removing line 541 will fix the problem

#### Version
master

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
all

#### Hardware type
all

#### Logs
N/A

rmackay9, 2020-02-05:
The code in question may not be intended to make the copter reduce it's altitude below the fence before returning but the intention may be to ensure that when the RTL_CLIMB_MIN and RTL_ALT are applied that the vehicle does not climb over the fence.  For example is CLIMB_MIN was 50m this could cause the vehicle to breach the fence even though it hadn't breached it before.

RhinoAus, 2020-02-05:
Interesting; I hadn't considered that case and I have never used RTL_CLIMB_MIN. I came at this from the perspective of the fence breach RTL behaviour not immediately attempting to resolving a vertical fence breach. I also don't fly with terrain so have not input there.