ArduPilot/ardupilot #13494: Arduplane: Throttle reversed in Loiter Mode

Avi8orn8, 2020-02-05:
### Bug report

**Issue details**
While in loiter and commanding a pitch/altitude change, throttle is reversed. 

_Please describe the problem_
When pushing forward on the elevator the aircraft will pitch down, and throttle will increase. When pulling back on the elevator the aircraft will pitch up, and throttle will decrease. Continue pulling back and the aircraft will eventually stall. Loiter works great if pitch/altitude change isn't commanded. 

**Version**
_What version was the issue encountered with_
Arduplane 4.0.3 (Chibios) and earlier versions

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Flying wing and V tail

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Matek F405 Wing, Pixracer, APM 2.8, Omnibus F4 Pro V2

**Logs**
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2020-02-06:
On Wed, 5 Feb 2020, Avi8orn8 wrote:

> When pushing forward on the elevator the aircraft will pitch down, and throttle will increase. When pulling back on the elevator the aircraft will pitch
> up, and throttle will decrease. Continue [...]

What were you expecting the pitch stick to do in loiter mode?

It does *not* change the loiter altitude.

I think what is happening here is that you are "stick mixing" the aircraft 
to be well-above where the loiter controllers want the aircraft to be. 
It's deciding it has way too much height, so cuts the throttle back and 
back and back to try to achieve its desired height.

If you check your logs, you should see TECS starting to get annoyed at the 
amount of energy it has and be pushing throttle down accordingly.

Avi8orn8, 2020-02-06:
Intuitively, I would expect pitch input to either control altitude in loiter mode similar to the way cruise mode responds to pitch input, OR do nothing at all. Pitching up/down without a logical response from throttle does not make sense, serves no purpose, and in fact could cause a crash if someone is not already familiar with this odd behavior.

As someone who uses automation on a routine basis in manned aircraft, having the ability to control altitude changes with pitch input makes the most sense to me, and would make loiter mode more useful in my opinion.

peterbarker, 2020-02-07:
On Thu, 6 Feb 2020, Avi8orn8 wrote:

> Intuitively, I would expect pitch input to either control altitude in loiter mode similar to the way cruise mode responds to pitch input, OR do nothing at
> all. Pitching up/down without a logical [...]

Yep, that's what I thought you were thinking - and I'm not saying you're 
wrong, just that you're thinking differently to the original author :-)

You might also have noticed that if you *stop* stick-mixing then the 
vehicle will bounce back to its originally loiter altitude.

In case you're wondering "why?!" - the idea behind stick-mixing is that 
the pilot may wish to temporarily push the aircraft.  A good example of 
this is on landing approach, where the autopilot doesn't always "get it 
just right", so some elevator or approach realignment is done.  The same 
logic is being used while in loiter mode - the aircraft just wants to get 
back "on track".

Having the option to make LOITER actually change alt on input may be 
acceptable to the Plane maintainers.  For that matter, moving the loiter 
point on roll/pitch input might also be acceptable.  That behaviour could 
be enabled by an RC channel option or perhaps with a parameter.