ArduPilot/ardupilot #13517: AP_Logger: should OneShot be logged in 1000~2000 range?

rmackay9, 2020-02-08:
There is some inconsistency in the ranges of outputs logged to the RCOU message depending upon the protocol used:

- Regular PWM: logs the actual PWM outputs
- OneShot: logs the actual PWM output which is in the range of 125 ~ 250
- DShot, ToshibaCAN, UAVCAN, KDECAN log in the range 1000 ~ 2000

The question is should we make OneShot act like Dshot & ToshibaCAN and log in the 1000~2000 range?  Alternatively should we make DShot, ToshibaCAN etc log using the "native" ranges that they send to their ESCs.  For example ToshibaCAN ESCs accept the speed input in the range 6300 to 32000.

Of course we all know that OneShot is just a special version of Regular PWM but this may not be how users look at it.  They don't think about the output so if we consider dataflash logs part of the user interface then keeping it consistent could reduce confusion and support calls.

This came up as part of this Copter-4.0 discussion where some users are struggling to setup DShot. https://discuss.ardupilot.org/t/pixhawk-4-holybro-arducopter-3-6-9-4-0-1rc3-no-control-over-two-motors-in-an-x8-with-dshot/51480

mlebret, 2020-02-09:
> OneShot: logs the actual PWM output which is in the range of 125 ~250 

OneShot125 do that. (125-250 range). OneShot may be in the 1000-2000 range if my understanding  is good.

Some pre-blheli32 esc (blheli-s) are dshot compatible. They were usable without restriction with arducopter 3.5.x. 

Since arducopter 3.6 we have to remap the PWM out of some cards, even to use these blheli-s ESC. Could we have an option to remain with the old RC out map with blheli-s (as with one shot).

Thanks for your dev work,

Marc

rmackay9, 2020-02-10:
We discussed this on the dev call and agreed that it would be very confusing if we logged the native outputs sent to the ESCs (see extreme example that would happen if we did this for ToshibaCAN ESCs).

We agreed that it would be fine to change OneShot125 to log "before the divide by 8 takes place" which would make the range used more consistent with regular PWM.

We also briefly discussed that it would be nice if we use floats in the 0 to 1 or 0-100 range instead but we judged this was a step to far at this stage.

PR #18591 (2021-09-08): AP_HAL_ChibiOS: convert back to full range on IOMCU ONESHOT125 read
Fixes #13517

Simply the inverse of the calculation done here:

https://github.com/ArduPilot/ardupilot/blob/0ee8b452ab76306d83c6252bcb9dc458bd7380bb/libraries/AP_HAL_ChibiOS/RCOutput.cpp#L497-L501

Tested on CubeOrange
closed 2021-09-08 by tridge