ArduPilot/ardupilot #13530: Too much telemetry data

cooldemo, 2020-02-09:
### Feature request

**Is your feature request related to a problem? Please describe.**
After upgrading to ArduPlane 4.0.3 my OSD and telemetry over 3G mysteriously stopped working. I traced it down to the amount of mavlink telemetry the ArduPlane produces. OSD is connected over 57.6k serial line and it is saturated. I did not have this problem with 3.9.x version.
OSD is MWOSD with default setting. 
For 3G telemetry the datarate is high even with every stream set to 1Hz. I think this must be happening to other radios too. 
Most of the values from telemetry are not needed every second. Imagine needing only 9.6k telemetry link :)

**Describe the solution you'd like**
My suggestion would be to:
- Increase granularity of telemetry selection to several bitfields (1 bit for every message type)
- Decrease minimum rate to less than 1Hz. (e.g. Negative numbers, -5 -> once every 5 seconds)

**Describe alternatives you've considered**
Alternatives are rising OSD UART speed to 115.2k and switching to 4G telemetry.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2020-02-09:
AP 4.0 supports the message interval command which allows much finer grain control of which messages the vehicle sends. I think it is probably mostly a ground station issue now in that the ground stations need to move from sending the request-data-stream messages to the message-interval messages.  That's not a super easy thing for the GCS developers to do but I think it is the next step.

FYI @meee1.

cooldemo, 2020-02-10:
Nice, I missed this.
Is this interface available in PARAMETERS for static configuration of intervals ?

rmackay9, 2020-02-10:
@cooldemo, no, I'm afraid the only way to make it stick on the autopilot side is to use a lua script ([example here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/message_interval.lua)).

There are two issues some will hit trying to use this Lua script:

- Lua takes quite a lot of memory meaning it barely fits on most flight controllers (it fits easily though on the CubeOrange and Holybro Durandal)
- you'll need to figure out the message numbers if you want to make more messages available.  this can be done by looking at [mavlink/common](https://mavlink.io/en/messages/common.html) but we should probably fill out the script with more of the messages that users will commonly want

peterbarker, 2024-07-31:
Modern ArduPilot has many (https://ardupilot.org/dev/docs/mavlink-requesting-data.html) ways of customising your stream rates should your bandwidth prove to be insufficient.

Closing this one now as its a touch old.

closed 2024-07-31 by peterbarker