ArduPilot/ardupilot #13532: Firmware.ardupilot.org: make it easier to determine if all binaries have been built

rmackay9, 2020-02-10:
It's quite difficult to determine if all binaries have been built by looking at [firmware.ardupilot.org](https://firmware.ardupilot.org/Copter/beta/).

For example I have [this support request](https://discuss.ardupilot.org/t/copter-4-0-2-rc4-available-for-beta-testing/52008/26) in which a user says they can't download the beta binary but the only way that I know of to independently determine if all the binaries have been updated is to go through each directory one-by-one.

In discussion with @tridge and @peterbarker recently we found that changing the web based file viewer back to the default allowed sorting the directories by update date which would (I think) allow me to determine if all binaries had been updated.

Pedals2Paddles, 2020-02-10:
Could whatever script is populating the files on the server create a running log that shows what it has spit out, preferably with most recent on top?  A CSV would be kind of useful and easy to analyze.

rmackay9, 2020-02-10:
Related issue here: https://github.com/ArduPilot/ardupilot/issues/13132

peterbarker, 2024-07-31:
@MichelleRos did this for us with `build_sizes.html` years ago,  Thanks Michelle!

closed 2024-07-31 by peterbarker