ArduPilot/ardupilot #13534: SYS_STATUS messages should include prearm status

RhinoAus, 2020-02-10:
### Feature request

**Is your feature request related to a problem? Please describe.**
Prearm checks passing is not reported by the SYS_STATUS message as described in the MAVLINK specification. Currently I use EKF GPS messages as the cue that my drone is ready to fly.

**Describe the solution you'd like**
Include the MAV_SYS_STATUS_PREARM_CHECK bit in the SYS_STATUS message

**Describe alternatives you've considered**
As far as I am aware, the only method for checking if the vehicle thinks its ready to arm is to wait for the "EKF is using GPS" messages and try arming it.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

oceanalan, 2020-11-03:
I think that this is a very useful feature.

I found that someone added this feature to the master branch on[ this commit ](https://github.com/ArduPilot/ardupilot/pull/13067) but it appears to have not made it into the Copter-4.0 branch

peterbarker, 2020-11-25:
We have this in master.

closed 2020-11-25 by peterbarker