ArduPilot/ardupilot #13544: Copter: MOT_PWM_MIN/MAX should be forced to 1000/2000 when using CAN ESCs

rmackay9, 2020-02-11:
When using non-PWM ESCs (BLHeli, ToshibaCAN, UAVCAN, KDECAN) the MOT_PWM_MIN/MAX values (or maybe SERVOx_MIN/MAX values?) must be set to 1000 and 2000 respectively or issues arise with the conversion of PWM to the other protocol.

We should consolidate the method we use to ensure the PWM range.  Currently we have these two methods:

- BLHeli forces the MOT_PWM_MIN/MAX values to 1000, 2000 at startup
- ToshibaCAN uses a pre-arm check to ensure they are set

ashvath100, 2020-02-19:
@rmackay9  should all the ESCs be consolidated to a pre-arm check approach or be forced at startup ?

rmackay9, 2020-02-20:
@ashvath100,

We should definitely force MOT_PWM_MIN/MAX to be 1000/2000 respectively if these smart ESCs are being used.  I think the new ESC_Telemetry library could be enhanced a bit so that callers do not need to directly check each of the individual libraries.

I would leave the pre-arm check in place for now as well though because it's always possible that users could change the MOT_PWM_MIN/MAX values after they've been automatically set.