ArduPilot/ardupilot #13546: Rename either SYSID_SW_MREV or FORMAT_VERSION parameter so they match

rmackay9, 2020-02-11:
The SYSID_SW_MREV and FORMAT_VERSION parameters are there to force parameters to be wiped when users load a different vehicle firmware to the autopilot.

We should consolidate the name of this parameter across vehicles so they are consistent

WickedShell, 2020-02-11:
Personally I find `SYSID_SW_MREV` to be a really confusing name for the param representing the internal storage format version number... Not saying `FORMAT_VERSION` is necessarily correct, I just think it's a lot less confusing then the other currently used one.

tridge, 2020-02-11:
i'd be happy with either one

rmackay9, 2020-02-11:
The nice thing about SYSID_ is that it hides the parameter in amongst other very rarely touched parameters while FORMAT_VERSION hangs out on it's own in the list.  I agree that it's not an easy to understand name though.

rmackay9, 2020-02-11:
The bad thing about SYSID_SW_MREV is that it's not a mavlink system id which all the other parameters are that start with SYSID_.  if we were picking the parameter name again I'd probably call it BRD_EEPROM_FMTID or something similar.

FORMAT_VERSION is also not a great name 'cuz it's missing the most important thing which is the system that it applies to, i.e. eeprom!

auturgy, 2020-02-11:
You might have answered your own question there Randy: new param name sounds sensible

rmackay9, 2020-02-11:
@auturgy, I suppose although I hate to make @meee1 handle yet another parameter rename.

peterbarker, 2020-02-11:
On Mon, 10 Feb 2020, Randy Mackay wrote:

> The bad thing about SYSID_SW_MREV is that it's not a mavlink system id which all the other parameters are that start with SYSID_. if we were picking the
> parameter name again I'd probably call it [...]

It probably *should* be BRD_ - but we store this value as a parameter 
rather than in any sort of magic.  We then format *all* of storage if it 
is different - not just the parameters but *everything*...  So yeah, it's 
all a bit strange....

Not sure what that means in terms of naming, 'though :-)

EEPROM_FMT_VER?

rmackay9, 2020-02-11:
@peterbarker, not a bad name but then what will we do if we move parameter storage to the SD card? :-)

PR #13600 (2020-02-19): Copter/Sub: Rename SYSID_SW_MREV to FORMAT_VERSION
This PR fixes #13546.
Please let me know if I missed something!
Thanks
closed 2020-02-19 by rmackay9