ArduPilot/ardupilot #13561: Missing Parameter Documentation

peterbarker, 2020-02-13:
### Bug report

@Hwurzburg noticed we were missing documentation for one of the Heli parameters.

I decided to write something to validate the SITL parameters, at least (PR: https://github.com/ArduPilot/ardupilot/pull/13560)

These results are from that branch.

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
SITL

**Logs**

The following parameters are missing documentation on a base `test.XXXX.SetHome` run of autotest with this validation running:

```
pbarker@bluebottle:~/rc/ardupilot(pr/missing-parameter-checks)$ ./Tools/autotest/autotest.py  --validate-parameters  fly.CopterAVC.SetHome
.
.
.
AUTOTEST: ---------- Validating parameters  ----------
AUTOTEST: -
Running: ("/home/pbarker/rc/ardupilot/Tools/autotest/param_metadata/param_parse.py" "--vehicle" "ArduCopter") in (/home/pbarker/rc/ardupilot)
AUTOTEST: Downloaded 1204 parameters OK
ATC_RAT_RLL_VFF not in XML
ATC_RAT_RLL_ILMI not in XML
ATC_RAT_PIT_VFF not in XML
ATC_RAT_PIT_ILMI not in XML
ATC_RAT_YAW_VFF not in XML
ATC_RAT_YAW_ILMI not in XML
PSC_ACCZ_FF not in XML
PSC_ACCZ_FLTT not in XML
PSC_ACCZ_FLTE not in XML
PSC_ACCZ_FLTD not in XML
>>>> FAILED STEP: fly.CopterAVC.SetHome at Thu Feb 13 14:32:23 2020 (Downloaded parameters missing in XML)
```

```
pbarker@bluebottle:~/rc/ardupilot(pr/missing-parameter-checks)$ ./Tools/autotest/autotest.py  --validate-parameters  build.APMrover2 drive.APMrover2.SetHome
.
.
SYSID_THISMAV not in XML
SPRAY_ENABLE not in XML
```
note extra underscore:     `// @Param: SYSID_THIS_MAV`

```
pbarker@bluebottle:~/rc/ardupilot(pr/missing-parameter-checks)$ ./Tools/autotest/autotest.py  --validate-parameters  build.ArduPlane fly.ArduPlane.SetHome
.
.
EFI_TYPE not in XML
LAND_DS_YAW_LIM not in XML
LAND_DS_L1_TCON not in XML
LAND_DS_P not in XML
LAND_DS_I not in XML
LAND_DS_D not in XML
LAND_DS_IMAX not in XML
LAND_DS_ABORTALT not in XML
```

```
pbarker@bluebottle:~/rc/ardupilot(pr/missing-parameter-checks)$ ./Tools/autotest/autotest.py  --validate-parameters  build.AntennaTracker test.AntennaTracker.SetHome
.
.
RC1_MIN not in XML
RC1_TRIM not in XML
RC1_MAX not in XML
RC1_REVERSED not in XML
RC1_DZ not in XML
RC2_MIN not in XML
RC2_TRIM not in XML
RC2_MAX not in XML
RC2_REVERSED not in XML
RC2_DZ not in XML
RC3_MIN not in XML
RC3_TRIM not in XML
RC3_MAX not in XML
RC3_REVERSED not in XML
RC3_DZ not in XML
RC4_MIN not in XML
RC4_TRIM not in XML
RC4_MAX not in XML
RC4_REVERSED not in XML
RC4_DZ not in XML
RC5_MIN not in XML
RC5_TRIM not in XML
RC5_MAX not in XML
RC5_REVERSED not in XML
RC5_DZ not in XML
RC6_MIN not in XML
RC6_TRIM not in XML
RC6_MAX not in XML
RC6_REVERSED not in XML
RC6_DZ not in XML
RC7_MIN not in XML
RC7_TRIM not in XML
RC7_MAX not in XML
RC7_REVERSED not in XML
RC7_DZ not in XML
RC8_MIN not in XML
RC8_TRIM not in XML
RC8_MAX not in XML
RC8_REVERSED not in XML
RC8_DZ not in XML
RC9_MIN not in XML
RC9_TRIM not in XML
RC9_MAX not in XML
RC9_REVERSED not in XML
RC9_DZ not in XML
RC10_MIN not in XML
RC10_TRIM not in XML
RC10_MAX not in XML
RC10_REVERSED not in XML
RC10_DZ not in XML
RC11_MIN not in XML
RC11_TRIM not in XML
RC11_MAX not in XML
RC11_REVERSED not in XML
RC11_DZ not in XML
RC12_MIN not in XML
RC12_TRIM not in XML
RC12_MAX not in XML
RC12_REVERSED not in XML
RC12_DZ not in XML
RC13_MIN not in XML
RC13_TRIM not in XML
RC13_MAX not in XML
RC13_REVERSED not in XML
RC13_DZ not in XML
RC14_MIN not in XML
RC14_TRIM not in XML
RC14_MAX not in XML
RC14_REVERSED not in XML
RC14_DZ not in XML
RC15_MIN not in XML
RC15_TRIM not in XML
RC15_MAX not in XML
RC15_REVERSED not in XML
RC15_DZ not in XML
RC16_MIN not in XML
RC16_TRIM not in XML
RC16_MAX not in XML
RC16_REVERSED not in XML
RC16_DZ not in XML
RC_OVERRIDE_TIME not in XML
RC_OPTIONS not in XML
SERVO1_MIN not in XML
SERVO1_MAX not in XML
SERVO1_TRIM not in XML
SERVO1_REVERSED not in XML
SERVO1_FUNCTION not in XML
SERVO2_MIN not in XML
SERVO2_MAX not in XML
SERVO2_TRIM not in XML
SERVO2_REVERSED not in XML
SERVO2_FUNCTION not in XML
SERVO3_MIN not in XML
SERVO3_MAX not in XML
SERVO3_TRIM not in XML
SERVO3_REVERSED not in XML
SERVO3_FUNCTION not in XML
SERVO4_MIN not in XML
SERVO4_MAX not in XML
SERVO4_TRIM not in XML
SERVO4_REVERSED not in XML
SERVO4_FUNCTION not in XML
SERVO5_MIN not in XML
SERVO5_MAX not in XML
SERVO5_TRIM not in XML
SERVO5_REVERSED not in XML
SERVO5_FUNCTION not in XML
SERVO6_MIN not in XML
SERVO6_MAX not in XML
SERVO6_TRIM not in XML
SERVO6_REVERSED not in XML
SERVO6_FUNCTION not in XML
SERVO7_MIN not in XML
SERVO7_MAX not in XML
SERVO7_TRIM not in XML
SERVO7_REVERSED not in XML
SERVO7_FUNCTION not in XML
SERVO8_MIN not in XML
SERVO8_MAX not in XML
SERVO8_TRIM not in XML
SERVO8_REVERSED not in XML
SERVO8_FUNCTION not in XML
SERVO9_MIN not in XML
SERVO9_MAX not in XML
SERVO9_TRIM not in XML
SERVO9_REVERSED not in XML
SERVO9_FUNCTION not in XML
SERVO10_MIN not in XML
SERVO10_MAX not in XML
SERVO10_TRIM not in XML
SERVO10_REVERSED not in XML
SERVO10_FUNCTION not in XML
SERVO11_MIN not in XML
SERVO11_MAX not in XML
SERVO11_TRIM not in XML
SERVO11_REVERSED not in XML
SERVO11_FUNCTION not in XML
SERVO12_MIN not in XML
SERVO12_MAX not in XML
SERVO12_TRIM not in XML
SERVO12_REVERSED not in XML
SERVO12_FUNCTION not in XML
SERVO13_MIN not in XML
SERVO13_MAX not in XML
SERVO13_TRIM not in XML
SERVO13_REVERSED not in XML
SERVO13_FUNCTION not in XML
SERVO14_MIN not in XML
SERVO14_MAX not in XML
SERVO14_TRIM not in XML
SERVO14_REVERSED not in XML
SERVO14_FUNCTION not in XML
SERVO15_MIN not in XML
SERVO15_MAX not in XML
SERVO15_TRIM not in XML
SERVO15_REVERSED not in XML
SERVO15_FUNCTION not in XML
SERVO16_MIN not in XML
SERVO16_MAX not in XML
SERVO16_TRIM not in XML
SERVO16_REVERSED not in XML
SERVO16_FUNCTION not in XML
SERVO_RATE not in XML
SERVO_VOLZ_MASK not in XML
SERVO_SBUS_RATE not in XML
SERVO_ROB_POSMIN not in XML
SERVO_ROB_POSMAX not in XML
PITCH2SRV_FF not in XML
PITCH2SRV_FLTT not in XML
PITCH2SRV_FLTE not in XML
PITCH2SRV_FLTD not in XML
YAW2SRV_FF not in XML
YAW2SRV_FLTT not in XML
YAW2SRV_FLTE not in XML
YAW2SRV_FLTD not in XML
STAT_BOOTCNT not in XML
STAT_FLTTIME not in XML
STAT_RUNTIME not in XML
STAT_RESET not in XML
```

```
pbarker@bluebottle:~/rc/ardupilot(pr/missing-parameter-checks)$ ./Tools/autotest/autotest.py  --validate-parameters  build.ArduSub dive.ArduSub.SetHome
.
.
PSC_ACCZ_FF not in XML
PSC_ACCZ_FLTT not in XML
PSC_ACCZ_FLTE not in XML
PSC_ACCZ_FLTD not in XML
```

rmackay9, 2020-02-13:
I've got fixes for a few of these in this branch: https://github.com/rmackay9/rmackay9-ardupilot/commits/psc-parm-fix.

@peterbarker maybe you could run this again on my branch to see if it's fixed the Sub, Rover, Copter and TradHeli issues?.. or I can just create a PR..

peterbarker, 2020-02-13:
On Thu, 13 Feb 2020, Randy Mackay wrote:

> @peterbarker maybe you could run this again on my branch to see if it's fixed the Sub, Rover, Copter and TradHeli issues?.. or I can just create a PR..

Sub: Fixed

Rover: Fixed

Copter: Fixed

TradHeli: Fixed

Plane: still broken :-)

Note that this is still only the "SetHome" test.  That means we're not 
hitting any "enabled" parameters with the tests I've run so far.

p.s. No love for Tracker? :-P

rmackay9, 2020-02-14:
OK, I've directly committed the fixes from my branch above to master..

peterbarker, 2020-02-19:
These were fixed.  Of course, now we've got the fact that the XML may contain parameters it shouldn't....

closed 2020-02-19 by peterbarker