ArduPilot/ardupilot #13563: WS2812B (NeoPixel) Notify Sync loss

IamPete1, 2020-02-13:
NeoPixels sometimes get out of sync with other notify LED's. They look to sometimes do not get the correct command but will re-sync later. This seems to be related to IO load, it is reported to get worse when more serial ports are used. More info here:

https://discuss.ardupilot.org/t/ws2812b-neopixel-integration/48580/56?u=iampete
https://drive.google.com/file/d/18cYoKydri3AWCNSzsWD8SwTqXV0C41gy/view?usp=drivesdk
https://discuss.ardupilot.org/t/ws2812b-neopixel-integration/48580/74?u=iampete

MallikarjunSE, 2020-02-13:
Hey @IamPete1 
I'm using Pixhawk 1 with fmuv2 firmware. My board is 1MB flash one. If you insist, I can try on CubeBlack or CubeOrange.
https://drive.google.com/open?id=12DvWVjWsykdbETWoXRXrgslGnfiLS377

MallikarjunSE, 2020-02-13:
If possible we should add NTF_NEO_LEN to upcoming version of Arducopter. This is only available in AC4.0.3dev

Another-Taipan, 2020-02-13:
I am using AP ver 4.0.3-dev on a Pixhawk 1 with fmuv3 firmware (a Pixhawk 2.4.8 with 2MB flash) and experience this same issue.

PR #13606 (2020-02-25): AP_HAL_ChibiOS: retry neopixel until successful send
Fixes #13563 This retrys sending neopixel until all groups get through. Tested on CubeOrange with lots of serial stuff , fixes sync loss.

Possibly more elegant to only retry failed groups, but this seems to work and is a small change.
closed 2020-02-25 by tridge