ArduPilot/ardupilot #13571: Access lock USB port(ttyACM0) to avoid firmware updates

CRAZYMONTY, 2020-02-14:
Hi everyone !
I am looking for a change to password protect or some kind of mechanism to block unauthorized access to ttyACM/USB port just to avoid any firmware updates to my Pixhawk.

peterbarker, 2020-02-14:
This is really an operating system question rather than autopilot question.

Somebody might be happy to help you out over on our discourse server anyway - http://discuss.ardupilot.org

We try to keep this issue list for verified problems with the ArduPilot software, so I'm closing this now.

auturgy, 2020-02-14:
I think there’s a little confusion: I’m reopening to clarify, but likely this is a support question so will be closed again and referred to the forum.
The USB port on the flight controller isn’t a ttyACM/USB, it just connects to one on a pc. I don’t think you want to lock your pc, but instead you want to lock the flight controller.
If that’s the case, close this and ask on discuss.ardupilot.org
If you actually do want to lock down the USB port on a pc, as Peter said, it’s an operating system question so we can’t really help.

CRAZYMONTY, 2020-02-15:
Okay got it, Thanks.

closed 2020-02-15 by CRAZYMONTY