ArduPilot/ardupilot #13591: Copter: allow setting RC failsafe timeout

rmackay9, 2020-02-17:
Copter currently uses a hard-coded RC failsafe timeout.  This timeout is 0.5sec for regular RC, 1sec for joysticks.  To view the code see [radio.cpp's read_radio method and search for FS_RADIO_RC_OVERRIDE_TIMEOUT_MS](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/radio.cpp#L111).

We should add a new parameter perhaps called FS_THR_TIMEOUT that allows a user to specify how long this timeout is.  I think probably the best way to do it is to have this parameter default to 0 in which case we use the existing timeouts, but if it's any other positive value then we use it.

This new parameter should be in seconds which is consistent with Rover's FS_TIMEOUT parameter and Plane's FS_SHORT/LONG_TIMEOUT parameters

peterbarker, 2020-02-18:
On Sun, 16 Feb 2020, Randy Mackay wrote:

> Copter currently uses a hard-coded RC failsafe timeout. This timeout is 0.5sec for regular RC, 1sec for joysticks. To view the code see radio.cpp's
> read_radio method and search for [...]

Shouldn't this go up into RC_Channels_VarInfo.h, right next to 
RC_OVERRIDE_TIMEOUT?

Rover could move to using it, and the other vehicles into the future.

Pedals2Paddles, 2020-02-18:
What happens during that timeout period?  When no pulses are received, does AP hold the inputs where they were at before they disappeared?  If the throttle channel drops below minimum for the low throttle method, what does the actual throttle do during the timeout? Ignore the low input and hold what it was at before, or cut off?

If AP holds the inputs where they were at, as in not crash-o-matic, then I think this could definitely be useful.  Operating at the fringe end of range can be very frustrating sometimes.

peterbarker, 2025-11-08:
Closed by https://github.com/ArduPilot/ardupilot/pull/22200