ArduPilot/ardupilot #13593: Plane: Add RCx_OPTION for Preflight Calibration

JoshWelsh, 2020-02-17:
Having "Preflight Calibration" be an option for RCx_Option would be awesome, we could then trigger this from the radio when ready rather than being required to invoke it from the GCS.

**Describe the solution you'd like**
Provide as one of the options available in RCx_Option

**Describe alternatives you've considered**

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**

peterbarker, 2020-02-18:
On Mon, 17 Feb 2020, Josh wrote:

> Having "Preflight Calibration" be an option for RCx_Option would be awesome, we could then trigger this from the radio when ready rather than being
> required to invoke it from the GCS.

Which preflight calibration?

Pedals2Paddles, 2020-02-18:
Also, why not call vehicles?  For example, gyro drift while warming up and baro drift with pressure is an all vehicles problem.

JoshWelsh, 2020-02-19:
@peterbarker the one I'm most specifically interested in is the one for the Airspeed sensor, but I don't think I was aware that there were more than one.  My request would be for the switch to do whatever the Preflight-Calibration does to the target vehicle.

@Pedals2Paddles Matt good call, I hadn't considered that.  I was focused primarily on that airspeed sensor calibration being performed while I had a cover on the pitot-static port, but I hadn't really thought far enough ahead to think about the gyros and baro drift.  Glad you brought that up!  I am editing the original post to comment on that.

EERomeo, 2020-02-20:
I had the same suggestion a few months ago, you can see here the whole discussion:
[https://github.com/ArduPilot/ardupilot/issues/12704](url)

IamPete1, 2021-08-16:
closing in favor of https://github.com/ArduPilot/ardupilot/issues/12704

closed 2021-08-16 by IamPete1