ArduPilot/ardupilot #13611: AP_RangeFinder: investigate and resolve RPLidar startup issue

rmackay9, 2020-02-19:
This reports comes from this discussion in the Rover-4.0 category: https://discuss.ardupilot.org/t/bad-lidar-health-message-with-rplidar-a2m8/52350

At least 3 users have reported that the RPLidar must be powered up before the flight controller or "Bad Lidar Health" messages appear on the HUD.

We should investigate and then try to resolve this issue.  Perhaps there are some startup messages sent by the driver that we should keep resending until we start receiving distances

andresrcdev, 2020-02-20:
The issue could be here in AP_Proximity_RPLidarA2.cpp

bool AP_Proximity_RPLidarA2::initialise()
{
    // initialise boundary
    init_boundary();

    if (!_initialised) {
        reset_rplidar();            // set to a known state
        Debug(1, "LIDAR initialised");
    }

    return true;
}

This returns _initialised=true, but it is not really checked that the lidar is working.

Also, following slamtec rplidar documentation, initialising scheme should be:
[image]

Let me check the code to see if I can help with something.

andresrcdev, 2020-02-27:
I have checked the code and I would like to make this suggestion to modify the initialise function.
The problem is that my coding skills are limited so I would need someone that could review the code and upload to test.
The AP_Proximity_RPLidarA2.cpp could be like that

```
#include <AP_HAL/AP_HAL.h>
#include "AP_Proximity_RPLidarA2.h"
#include <AP_SerialManager/AP_SerialManager.h>
#include <ctype.h>
#include <stdio.h>

#define RP_DEBUG_LEVEL 0

#if RP_DEBUG_LEVEL
  #include <GCS_MAVLink/GCS.h>
  #define Debug(level, fmt, args ...)  do { if (level <= RP_DEBUG_LEVEL) { gcs().send_text(MAV_SEVERITY_INFO, fmt, ## args); } } while (0)
#else
  #define Debug(level, fmt, args ...)
#endif

#define COMM_ACTIVITY_TIMEOUT_MS        200
#define RESET_RPA2_WAIT_MS              2                   // Acording to RPLidar A2 Documentation
#define RESYNC_TIMEOUT                  5000

// Commands
//-----------------------------------------

// Commands without payload and response
#define RPLIDAR_PREAMBLE               0xA5
#define RPLIDAR_CMD_STOP               0x25
#define RPLIDAR_CMD_SCAN               0x20
#define RPLIDAR_CMD_FORCE_SCAN         0x21
#define RPLIDAR_CMD_RESET              0x40

// Commands without payload but have response
#define RPLIDAR_CMD_GET_DEVICE_INFO    0x50
#define RPLIDAR_CMD_GET_DEVICE_HEALTH  0x52

// Commands with payload and have response
#define RPLIDAR_CMD_EXPRESS_SCAN       0x82

extern const AP_HAL::HAL& hal;

/*
   The constructor also initialises the proximity sensor. Note that this
   constructor is not called until detect() returns true, so we
   already know that we should setup the proximity sensor
 */
AP_Proximity_RPLidarA2::AP_Proximity_RPLidarA2(
    AP_Proximity &_frontend,
    AP_Proximity::Proximity_State &_state) :
    AP_Proximity_Backend(_frontend, _state)
{
    const AP_SerialManager &serial_manager = AP::serialmanager();
    _uart = serial_manager.find_serial(AP_SerialManager::SerialProtocol_Lidar360, 0);
    if (_uart != nullptr) {
        _uart->begin(serial_manager.find_baudrate(AP_SerialManager::SerialProtocol_Lidar360, 0));
    }
    _cnt = 0;
    _cnt_resetted = 0 ;
    _sync_error = 0 ;
    _byte_count = 0;
    _rplidar_mech_error_cnt = 0;
    _rplidar_comm_error_cnt = 0;
    _rplidarOk = false;
    _sector_initialised = false;
    _initialised = false;
    _rp_state = rp_unknown;

}
[... 407 lines cut ...]
                _sync_error++;
            }
            break;

        case ResponseType_Health:
            // health issue if status is "2" ->HW error
            if (payload.sensor_health.status == 2) {
                Debug(1, "LIDAR Error");
                _rplidar_mech_error_cnt ++;
                _rplidarOk = false;

            } 
            if (payload.sensor_health.status < 2) {
                _rplidarOk = true;
            }
            break;

        default:
            // no valid payload packets recognized: return payload data=0
            Debug(1, "Unknown LIDAR packet");
            break;
    }
}
```
And also the corresponding changes to the header file.

amilcarlucas, 2020-05-11:
Can you please raise a github PR with that

andresrcdev, 2020-05-12:
Yes, but I am very bussy at work, so it won't be possible before weekend.
I think I will also need some help.

Arduhawk, 2020-08-10:
What is the status on this? I am having troubles even running the pixhawk on a separate circuit and plugging power into the pixhawk after the lidar has spun up.

acobo, 2022-02-28:
Hello all,
 I'm having problems with a RPlidar A2M8 and Rover 4.1.5, half of the times the lidar is not detected and the "bad lidar health" message is shown. I guess it is due to this initialization problem. 
 Do anyone know the status of this proposed fix in the code?
 I am reading posts suggesting that the lidar should be switched on before the pixhawk, other posts suggest the opposite solution, but in  any case, it involves additional hardware to do that. 
Thankyou in advance!
 Adolfo.