ArduPilot/ardupilot #13615: Crash/Watchdog reset on 1st time rc input receiving

Jaaaky, 2020-02-19:
## Bug report

**Issue details**

I tested the following on pixhawk1 but I'm pretty sure it affects others boards too.
In summary, with signing enabled, receiving RC-Input causes a crash!

The probability of the problem increases:
1. if you increase the optimization level from Os to O2 or O3
2. with higher SCHED_LOOP_RATE
_But it occurs with everything at default too_

Steps to reproduces:

1.  Add env OPTIMIZE -O3 to hwdef.data - you can skip this it just increases the probability
2.  build and upload
3.  using mavproxy run `signing setup blabla`
4.  `reboot` the board
5.  wait until long failsafe is triggered and RTL is active - not necessary but it increases the probability of getting the problem.
6.  Connect any SBUS RC receiver into RC-IN port. Actually it could already been connected with the transmitter turned-off then turned-on at this point.
7. FMU WATCHDOG reset would be triggered now over 50% of times. With O3 it can randomly completely freeze and the watchdog is dead :-( - power cycle would be needed.

It occurs only once per boot, so if you diconnect and reconnect the reciever over the same boot session it doesn't re-occur.
But if the RC transmitter was not turned on like in AUTO takeoff for example, then turned-on , it would cause in-flight watchdog reset and possibly crash.

**On ArduPlane;** the crash occurs after message "APM: RCInput: decoding SBUS" .. Before it says "Throttle failsafe off" ..
Increasing scheduler loop rate from 50 to 400 increases the chance the crash occurs

**On ArduCopter;** it occurs almost on every boot with everything at default other than enable signing.  Maybe because copter has high SCHED_LOOP_RATE by default

By the way I'm pretty much sure it occurs in other cases too not related to signing, but I couldn't reproduce other cases

**Version**
_mastere_

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk1, I think all fmuv3 are affected, maybe all ChibiOS boards.

rmackay9, 2020-02-19:
Perhaps it's a coincidence but this report in the forums has some of the same keywords: https://discuss.ardupilot.org/t/crash-following-rcinput-decoding-sbus/52290

Jaaaky, 2020-02-19:
@rmackay9 Oh yeah awful, I've already reported this on gitter/ChibiOS few months ago, but the MON log didn't have any useful info, - another bug - . But I couldn't reproduce it until today. 
I'm trying to get into the root cause, maybe you can help bisecting with me.

Jaaaky, 2020-02-19:
Thanks God, I could bisect this very stupid bug. 
first bad commit: [7168a0297aa346d461593eed9073512dd411dbfc] HAL_ChibiOS: display RC protocol for iocmu

 I'm posting a quick fix now, until the GCS stack is properly fixed.

tridge, 2020-02-19:
thanks, I've reproduced the issue. I'll track down cause now

tridge, 2020-02-19:
the cause is that the RCIN thread runs out of stack due to the extra stack used when signing packets. I'll do a fix now

tridge, 2020-02-19:
having signing enabled adds 576 bytes to stack cost of send_text(). Tracking down why that is so high now

rmackay9, 2020-02-20:
This is fixed in master so closing.  It's going out in Copter-4.0.3-rc1 today most likely.

Happy to re-open if people want to of course.  thanks again for this!

closed 2020-02-20 by rmackay9