ArduPilot/ardupilot #13618: IO processor causes motor spool-up on aircraft with reverse thrust

SUI-Eric, 2020-02-19:
## Bug report

**Issue details**

In situations where the IO controller is no longer communicating with the FMU (soft reboot, firmware update, etc.) some situations seem to exist where the IO controller asserts low PWM on the throttle output channels(s). In most situations this would not cause issues. However, it can result in unintended spool-ups to full power in one direction on aircraft configured for reverse thrust.

**Version**
Plane 3.9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Hitec Xeno FX

**Hardware type**
Cube Black

**Logs**
https://youtu.be/ZBMU6mb1TIQ

peterbarker, 2020-02-20:
On Wed, 19 Feb 2020, Eric Maglio wrote:

> In situations where the IO controller is no longer communicating with the FMU (soft reboot, firmware update, etc.) some situations seem to exist where the
> IO controller asserts low PWM on the [...]

Telemetry and dataflash logs, please.

dataflash logs with LOG_DISARMED set if required.

What was happening with RC input during this test?

JoshWelsh, 2020-02-20:
Curious, what will the logs show if the FMU isn’t booted yet?

peterbarker, 2020-02-20:
@JoshWelsh it will show us the RC input states immediately before and immediately after the reboot, at the very least.  Armed status also....

SUI-Eric, 2020-02-20:
Ok, this should be the relevant tlog from the test during which the video was filmed. I can try and fetch  a dataflash log if it'd help. 

https://www.dropbox.com/s/n6dnr10bp4p9jz6/2020-02-04%2009-39-45.tlog?dl=0

First instance of the issue happens about 24% through the log. It's worth mentioning that we've modified the firmware on our handheld controller such that 1500us is the lowest value such that it's not possible to command reverse thrust with the stick even in manual mode.

peterbarker, 2020-02-20:
On Wed, 19 Feb 2020, Eric Maglio wrote:

> First instance of the issue happens about 24% through the log. It's worth mentioning that we've modified the firmware on our handheld controller such that
> 1500us is the lowest value such that it's not possible to command reverse thrust with the stick even in manual mode.

I wonder if this will persist for as long as the vehicle is in the 
"initialising" state....

If you could try to keep it in the initialising state by moving the 
vehicle around during boot - does it keep the motor running for a 
significant period of time?  Might pay to remove any prop for this test 
;-)

I do note that you're using PX4IO_SPEKTRUM; there's special handling for 
that in the iofirmware code.  Do you have the option of trying a different 
receiver to see if you can reproduce?

SUI-Eric, 2020-02-20:
Yes, we'll get the aircraft into the error state to test both of those ideas and report back.

Receiver is a customized Multiplex model running SRXL protocol, but we can swap in a PPM one to test.