ArduPilot/ardupilot #13625: Plane: Poor multi-link telemetry performance after upgrade from plane 3.9 to 4.0

MartinSollie, 2020-02-20:
**Issue details**
The message rate and usability of multiple links appear to have become significantly worse after upgrading from plane 3.9 to 4.0.

Setup details: We are using a Pixhawk 1 with serial 1 connected to a SiK radio (57600 baud), serial 2 connected to an onboard Odroid XU4 (921600 baud) and serial 4 connected to a router used for an Ubiquiti Rocket M5 radio (115200 baud). 
On the ground we are using another SiK radio and a Rocket M5 connected to Mavproxy, which is then connected to MissionPlanner using TCP. The onboard Odroid sends commands to stream position, attitude and more at 25Hz.

With plane 3.9.11, this setup works alright for all links, although scheduling problems is apparent in the 25Hz position data, as the time between messages often increase with intervals corresponding to 1 loop (0.02s at 50Hz default loop rate), such that the time between messages regularly increase to 0.04, 0.06, 0.08s and such. With plane 4.0.3, this data drops to a rate of only ~2.5 Hz, but often has dropouts for several seconds (see plots below). Another way this problem is apparent is when loading parameters after connecting to MissionPlanner. With 4.0.3, this sometimes stops early and does not finish, or is just slower than usual. This has not occured with 3.9 for us.

Increasing SCHED_LOOP_RATE from the default 50 to 400 (default for copter) solves the problem, and all links work well. This parameter however cannot simply be increased as it also affects other things such as flight control, but probably means that copter users will not encounter the problem.

Today we also tested this with a Durandal, which works a lot better, although this still shows improvements when increasing SCHED_LOOP_RATE.

The problem is most likely related to https://github.com/ArduPilot/ardupilot/issues/13530 and https://github.com/ArduPilot/ardupilot/issues/12077#issuecomment-523222846, and the problems described on the forum [here](https://discuss.ardupilot.org/t/telem1-vs-telem2-getting-messages-from-pixhawk/51789).

Here are some plots showing the time between position messages, requested at 25Hz (so ideally the difference in timestamp should be 0.04s), from the onboard computer. Using the timestamps from this will introduce some jitter, but the difference is still very visible.
[image]
[image]
[image]
[image]
[image]

**Version**
_What version was the issue encountered with_
ArduPlane 4.0.3 (but I suspect this problem was introduced in the first 4.0.0 beta).

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Does not matter, this issue is visible also on ground without flying.

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk 1, Durandal

**Logs**
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2020-03-10:
This may have been fixed with https://github.com/ArduPilot/ardupilot/pull/13662/commits

This did not make the Plane 4.0.5 stable, but it would be good to know if it fixes your issues as we can then stop looking for more problems.

peterbarker, 2020-06-19:
@MartinSollie there's a Plane beta out which contains a fix for poor telemetry performance.  Is it easy for you to check to see if it fixes the problem you've been seeing?

IamPete1, 2021-07-26:
No update from @MartinSollie, @peterbarker thinks its fixed.

closed 2021-07-26 by IamPete1