ArduPilot/ardupilot #13647: [Trad-Heli]Throttle curve problem

jinchengde, 2020-02-22:
## Bug report

**Issue details**
trad-heli
H_RSC_MODE : 3(throttle curve)
H_RSC_THRCRV_0 : 0
H_RSC_THRCRV_25 : 25
H_RSC_THRCRV_50 : 50
H_RSC_THRCRV_75 : 75
H_RSC_THRCRV_100 : 100
IM_STB_COL_1: 40
IM_STB_COL_2: 60
IM_STB_COL_3: 70
IM_STB_COL_4: 90

when arm the vehicle and enable the interlock, rc 8 output should stay at 1100 becuase H_RSC_THRCRV_0 is 0, but it raise up to larger than 1400, it looks some conflict with IM_STB_COL parameter because IM_STB_COL_1 is set as 40.

the video is below
https://www.youtube.com/watch?v=p9WsfJZHiss&feature=youtu.be

**Version**
4.0.4 dev

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
helicopter

**Hardware type**
sitl, pixhawk4-mini

**Logs**
_Please provide a link to any relevant logs that show the issue_

jinchengde, 2020-02-27:
so could we change the im_stb_xxx from 4 points to 5 points to match the throttle curve?
@bnsgeyer @MidwestAire

MidwestAire, 2020-02-27:
IMO the IM_STAB_COL should be removed from the code. It does not work with acro, and its only purpose is to line up the collective stock in manual modes with center for altitude controlled modes. This is easier achieved, much more accurate and better results, using a switched-in collective curve in the RC, switched in with the flight mode switch, than it is doing it in software. And even the most primitive of RC radios can do it. See this thread:

https://discuss.ardupilot.org/t/acro-collective-to-match-stab-collective/42087

bnsgeyer, 2020-02-27:
@jinchengde There is no problem with the code.  It works as designed.  IM_STB_COL parameters allow the user to set a collective curve for the stabilize flight mode.  It appears that you have a 3D setup where in acro mode you have a large negative blade collective.   As I'm sure you know, the limits of the acro blade collective pitch are set with the H_COL_MIN and H_COL_MAX parameters.  Then in stabilize, it appears that the blade collective minimum is some small negative value which is set up with the IM_STB_COL_1 and the maximum blade collective pitch is set with IM_STB_COL_4.  So if you are in stabilize and arm and enable motor interlock, then the HeliRSC servo output will go to the point on the throttle curve that is associated with the blade collective pitch given by the stabilize collective curve.  The throttle curve is based on the full collective blade pitch range set by H_COL_MIN and H_COL_MAX. 
 In RSC Mode 3, throttle curve has nothing to do with the RC8 output.  It uses the RC transmitter throttle stick position to determine, based on the manual flight mode (stabilize or acro), what the blade collective pitch is and then determines the HELIRSC servo position using the throttle curve.

So are you using an electric motor or a gas engine?  I don't understand why your throttle curve is a straight line from 0 to 100 for the output.

Jin, please take this conversation to an ardupilot discuss post.  We can work through some of the issues you are experiencing in your setup.  If you like you can open an issue in github for an enhancement to the tradheli code for the 5 point stabilize collective curve.  I would like to close this issue since I believe this is more of a user setup problem.

MidwestAire, 2020-02-27:
@bnsgeyer while the setup demo'd is indeed not flyable, the Input Manager is still problematic. It's design never considered that a fuel helicopter will take off at a greater weight than what it lands at after an hour of flight time. And the collective position is then different at landing than it was at takeoff.

Plus it doesn't work with acro. So it is a limited-use system, applies to only fixed-weight machines. This is easily set up and compensated for in a RC setup. And this setting can't be used unless RC radio is being used in the first place. So why not transfer something that doesn't work right, except for a limited use-case, to a control system that is better able to handle it and is virtually infinitely more configurable?

bnsgeyer, 2020-02-27:
@MidwestAire  Yep got it.  I agree that the Input manager is not ideal, however the work around to doing it your way in the interim is to set the IM_STB_COL for full range.  Then the RC radio can be used to configure the collective curves based on flight mode.  Is that not the work around or am I missing something.

MidwestAire, 2020-02-27:
@bnsgeyer that's exactly what we're doing at present. Works in both acro and stabilize, loaded or unloaded. Just have to set up the curves for it the RC and they are flipped in or out with the flight mode switches automatically. In training the pilots will learn about selecting the loaded vs unloaded curves. When the helicopter comes in light after a long flight it will suddenly climb when switched from Pos Hold to acro if the loaded curve is used. That mistake is usually only made once and then they learn that the landing checklist is just as important as the takeoff checklist.

What's in the input manager was designed for fixed-weight electric, with no consideration given to acro at all. I think it was done back in the day to "fix" setting the COL_MID to hover collective to get things to line up with the altitude modes. I believe it appeared in 3.3.3. It was probably tested with a Trex 450 or similar, "it works", it's been that way ever since. But it only works for a narrow use-case.

jinchengde, 2020-02-28:
@bnsgeyer 
the problem is that most user will confused by the IM_STAB_COL_XXX and H_RSC_THRCRV_XXX, take example, set parameter as:
IM_STB_COL_1: 40
H_RSC_THRCRV_0 : 0
RSC Mode: 3
then you are in stabilize, RC channel 3 is in lowest position, and arm and enable motor interlock, in normal, the throttle output is still keep in 1100 because you rc 3 input is in lowest and throttle curve is from 0, but in currently , the throttle output will increase , you could see it in demo before.
as in normal, the throttle curve and collective curve is two different system, they all relation with the rc3 input

MidwestAire, 2020-02-28:
No users are getting confused. If you have a large negative pitch setup and set IM_STB_COL_1 to 40 then that is for Stabilize. The setup would have to mean that 40% collective position is somewhere around feather pitch or you can't even start up the heli. At the same time, the throttle curve positions for 25 and 50% would be set to correspond to the correct power for that collective position, no matter what mode you are in. You should be running a V throttle curve. This is Basic Heli Setup 101 for many, many years.

So there is no problem other than a deliberately devised wrong setup.

Further it remains to be explained why anyone would run large negative pitch with an autopilot setup. The system is not capable of hard 3D flight. And the basic sport aerobatics the system is capable of are all positive-G maneuvers. Meaning more negative pitch than you need for autorotation is is totally not even useful.

jinchengde, 2020-02-28:
thanks chris，
sorry I have some mistake, I set the **IM_STAB_COL_0** to 40 for stablize.

and in my side, I always set H_COL_MIN to -12, and H_COL_MAX to +12, then adjust the IM_STAB_COL_0 to 40, IM_STAB_COL_4 to 90, then the minum collective will go to -3 and maxium collective will go to +8, as you said , that is wrong setup, even through the collective is the same for used but the throttle curve will influence.

so for noraml uav heli, IM_STAB_COL_0 should be set 0, and decrease the H_COL_MIN to small negative pitch and IM_STAB_COL_4 should be set 100 to match the H_COL_MIN/H_COL_MAX

MidwestAire, 2020-02-28:
The setup of your pitch range and use the IM_STAB_COL is not an issue at all. The problem is that your throttle curve has to be tuned to match that pitch range. Which is a V throttle curve. Then it will work just fine. The throttle curve always has to be tuned to match the power requirement.

The reason the IM_STAB_COL has four settings is because it allows for "flattening" of the collective curve in stabilize at center stick. If it had an odd number of points, such as five, the center point, #3, would have to be exactly the average of points 2 and 4 or you would get an un-flyable collective curve.

bnsgeyer, 2020-02-28:
@jinchengde there is no bug in the software so I am closing this issue. Looks like you brought it up on the discuss site.

closed 2020-02-28 by bnsgeyer

jinchengde, 2020-02-29:
thanks @bnsgeyer