ArduPilot/ardupilot #13663: Rover: send text warning to user if they try to calibrate accels or compass while armed

rmackay9, 2020-02-25:
If a user tries to start an accelerometer or compass calibration while the vehicle is disarmed we should send a warning message to the user to tell them why the calibration won't start.

This is especially important on Rover which allows the user to disable the requirement for arming.  TBH, I'm not a fan of being able to disable arming but if we're going to continue to allow this then we should send a warning when the calibration can't be started

This issues is raised because of this discussion in the forums: https://discuss.ardupilot.org/t/here2-here-compass-problems-solved/52715

rishabsingh3003, 2020-02-25:
Hi! Please assign this issue to me. I'd like to submit a patch for it. 
If I understand correctly, `gcs().send_text` is to be used to alert the user at GCS, in case they try to calibrate after arming the vehicle?
Thanks

rmackay9, 2020-02-25:
@rishabsingh3003,

Yes, that's right.  Somewhere I think there should be somewhere in the compass calibration and accel calibration code that checks if the vehicle is soft armed.  Wherever that check is we should add a sent-text I think.

PR #13667 (2020-03-03): Send text warning to user if they try to calibrate accels or compass while armed
This PR fixes #13663
With this PR GCS will be notified if Calibration is done while the vehicle is armed.
MAV_SEVERITY_ALERT seemed appropriate to me (although we can possibly use warning/critical)?
Please let me know if any changes are required.
closed 2020-03-03 by tridge