ArduPilot/ardupilot #13665: Plane: integrate AP_LandingGear's new OPTIONS parameters

rmackay9, 2020-02-25:
This PR https://github.com/ArduPilot/ardupilot/pull/13638 enhances the AP_LandingGear library by adding a new OPTIONS parameter so users can specify if they would like the landing gear to automatically retract after takeoff and/or deplaoy during landing.  We should integrate this into Plane and QuadPlane as well.

rmackay9, 2020-03-04:
I had a serious look at trying to implement this for plane but it's beyond me.  One change that I think would be good is to add these lines to the bottom of [Plane::complete_auto_takeoff()](https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/takeoff.cpp#L269):

>    // optionally retract landing gear
>    g2.landing_gear.retract_after_takeoff();

tridge, 2020-04-01:
fixed in #13953

closed 2020-04-01 by tridge

pompecukor, 2020-04-14:
@tridge @rmackay9 

It is still not working, testing with options set to 0 or to only land. And it would still retract on takeoff.

In the below picture retract is set to 65m, Transition is set to 30m. Option is set to 0.
As you can see it still retracts.

[image]