ArduPilot/ardupilot #13678: "HAL_BOARD_SUBTYPE_CHIBIOS_OMNIBUSNANOV6" is not defined

peterbarker, 2020-02-27:
## Bug report

**Issue details**

The board subtype being used on omibusnanov6's hwdef file doesn't actually exist.

```
BB-WAF: ../../../libraries/AP_RangeFinder/AP_RangeFinder.cpp:430:6: note: in expansion of macro 'CONFIG_HAL_BOARD_SUBTYPE'
BB-WAF:  #if (CONFIG_HAL_BOARD_SUBTYPE == HAL_BOARD_SUBTYPE_LINUX_BEBOP || \   
BB-WAF:       ^
BB-WAF: ./hwdef.h:36:34: warning: "HAL_BOARD_SUBTYPE_CHIBIOS_OMNIBUSNANOV6" is not defined [-Wundef]
```

**Version**
master

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n/a

**Hardware type**
omnibusnanov6

**Logs**
[above]

andyp1per, 2020-02-27:
Blast from the past:

https://github.com/ArduPilot/ardupilot/pull/12800 <https://github.com/ArduPilot/ardupilot/pull/12800>

Andy

peterbarker, 2020-03-03:
Fix is incorrect; remove the define of this symbol in the relevant hwdef instead of adding it to the enumeration.

andyp1per, 2020-03-03:
Ok, just skiing ATM ...

peterbarker, 2020-08-09:
This issue was fixed

closed 2020-08-09 by peterbarker