ArduPilot/ardupilot #13687: Capability of SLEMTECH S1 Lidar (360 degree)

ziniu888, 2020-02-28:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request
**Capability of SLEMTECH S1 Lidar (360 degree)**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
**The S1 lidar is veery similar to RPLidar A2 360 degree lidar. The S1 lidar can get distance of 40 meters (twice than A2) and light-weight of 100g (almost a half weight of A2).**

There might be few lines of code need to be changed!

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[√] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

EosBandi, 2020-06-03:
I just got my S1, the only difference is that it does not send out anything after reset. So ideally the change in the driver is only a couple lines of code (if keep using it in compatibility mode). Will do a flight test in the upcoming week(s) and then check in the changes to the driver.

EosBandi, 2020-06-05:
PR added https://github.com/ArduPilot/ardupilot/pull/14527

rmackay9, 2023-11-28:
ArduPilot 4.5 includes support for the S1 (see PR https://github.com/ArduPilot/ardupilot/pull/23749) so I'll close this, thanks!

closed 2023-11-28 by rmackay9