ArduPilot/ardupilot #13689: [SERIOUS] EKF screwed in dual GPS setup, one of which is an RTK assisted Emlid Reach

GlidLov, 2020-02-28:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

in a copter setup (Cube 2.1) which has two GPS units, one is an Here2 (M8P) the other is the Emlid Reach M+  plus the Reach RS+ on ground.

When system is configured in "blend" GPS mode (GPS_AUTO_SWITCH = 2) and base is not turned on, copter flies OK. If you turn on the base, and it begins to comunicate with Rover (airborne), EKF go mad, copter lose positioning and if you're in a GPS assisted flight mode, copter flies away.

If you switch off ground base, problem does not occur. 
If gps auto switch is disable, so it's only using first GPS unit, problem does not occur.(GPS_AUTO_SWITCH = 0)  even if base is on and connected.

I think blending alghorithm does not handle this kind of mixed setup, maybe it thinks that a poor (lower sats) fix from a "rtk" GPS source is given priority towards a non-RTK one which has a better fix, instead.

**Version**
Arducopter 3.6.9 

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

proficnc, 2020-02-28:
Blend must be matching GPS’s 

Please post these kinds of questions on the forums, that’s the best place for help.

closed 2020-02-28 by proficnc

auturgy, 2020-02-28:
That’s quite an old version of copter, so first step is seeing if this behaviour remains in the current 4.0 releases,  then if it does, raise this in the forum with logs.

Regards,

James

GlidLov, 2020-07-17:
updated to 4.0.2, with NO gps  blending, bug is still there. quad use also a lightware rangefineder on the front as proximity/avoidance. If this one is active AND emlid base RTK is on and active, EKF goes crazy. If i disable rangefinder OR turn off base, it goes OK.

GlidLov, 2020-07-17:
> ## Bug report
> in a copter setup (Cube 2.1) which has two GPS units, one is an Here2 (M8P) the other is the Emlid Reach M+ plus the Reach RS+ on ground.
>
> When system is configured in "blend" GPS mode [...]

auturgy, 2020-07-17:
Please raise this in the forum (discuss.ardupilot.org) with logs.

rmackay9, 2020-07-17:
@GlidLov, Txs for re-testing with 4.0.2.  It is odd that the rangefinder has any impact because the EKF shouldn't be using the rangefinder.  I wonder if perhaps EKx_ALT_SOURCE = 1 (rangefinder) which would be bad especially in outdoor environments.  Feel free to post a link to the discussion here when you open it and I'll try and have a look.