ArduPilot/ardupilot #13697: error: 'hal' was declared 'extern' and later 'static'

8-DK, 2020-02-29:
## Bug report

**File cause this issue**
libraries/AP_HAL_ChibiOS/Device.cpp:34:27: 

**Issue details**
Compiling code in eclipse give the error
`[542/719] Compiling libraries/AP_HAL_ChibiOS/Device.cpp
../../libraries/AP_HAL_ChibiOS/Device.cpp:34:27: error: 'hal' was declared 'extern' and later 'static' [-fpermissive]
 static const AP_HAL::HAL &hal = AP_HAL::get_HAL();
                           ^~~
compilation terminated due to -Wfatal-errors.`

**Version**
Copter-4.0.2 (aae944a)

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
fmuv2

**Changes suggested**
` extern const AP_HAL::HAL& hal;`

peterbarker, 2020-02-29:
@8-DK I was unable to replicate that locally here.

It's a bit unusual, 'though, so I made up https://github.com/ArduPilot/ardupilot/pull/13698

Any chance you could test that PR?

peterbarker, 2020-11-25:
Closing this as no response and the PR was merged.

closed 2020-11-25 by peterbarker