ArduPilot/ardupilot #13709: Copter: extra channel appearing in RCIN

rmackay9, 2020-03-02:
There have been at least two reported cases (over a long period of time) in which the RC input library has reported one more channel than actually exists on the transmitter.

@R-Lefebvre has seen this with SBU
@rmackay9 has a log from a user who was using PPM input to the autopilot.  This PPM signal came from an RFD900x radio that was connected to a custom GCS on the ground.  It's unclear at this point if the extra channel was temporarily created by the GCS softare, during the transmission over the RFD900x or was a result of some issue in AP's decoding of the signal

In the case reported by rmackay9, the RCIN logging showed 12 channels for 99.9% of messages, but occasionally (about 3 times over 2.5hrs) it reported a 13th channel.  This channel was actually channel 12 and it appeared that channel 3 or channel 4 had been duplicated and then all subsequent channels were shifted along one position.

davidbuzz, 2020-03-10:
following

rmackay9, 2020-04-03:
I think we have concluded that this is an issue with the RFD900x.  It would be best if the fixed version of the RFD900x was tested and proven to be the cure but in any case, it seems not to be an ArduPilot issue so I'll close this.  All comments welcome though.

closed 2020-04-03 by rmackay9

rmackay9, 2020-05-26:
RFDesign have apparently included the fix in the RFD900 firmware version V3.15 (i.e. RFDesign firmware, not ArduPilot).