ArduPilot/ardupilot #13732: SITL: Empty/Missing eeprom.bin and default paramater file causes an invalid eeprom.bin to be generated.

matthewmcgill, 2020-03-05:
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

When starting a new SITL simulator instance I do so in an empty directory and use a default parameters file.  The SITL the creates a new eeprom.bin but does not add the parameter FORMAT_VERSION to the eeprom file (Most likely because it is the default value).  This prevents the reading of parameters from the eeprom file on subsequent invocations.

If run without a default parameters file FORMAT_VERSION is added to the eeprom.bin allowing parameters to be read/written to as normal on later executions.

Using a valid(including FORMAT_VERSION) eeprom.bin with a default parameters file also works as expected.

**Version**
Currently branched off master(3194059408a885a19b74e7ff6cb1350cdd2c50d6)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Airframe type**
SITL, boat/rover

**Hardware type**
SITL

**Logs**

khancyr, 2020-03-05:
hello, 

I tried to reproduce your issue but SITL is working well. Are you sure that you aren't passing the -w parameter ?

khancyr, 2020-03-05:
do you have step to reproduce the issue ?

matthewmcgill, 2020-03-06:
The bug i believe is in AP_Param, but as only SITL really uses a defaults file, I didn't know how to allocate it.

Steps to reproduce:
  1) checkout master
  2) ./waf configure --sitl
  3) ./waf rover
  4) create an empty simulation director, cd to it
  5) ./ardurover -M rover --defaults example_defaults.parm
  6) connect to the simulated rover
  7) change any parameter (ie: SIM_WIND_SPD)
  8) stop ardurover
  9) ./ardurover -M rover --defaults example_defaults.txt
  10) connect to rover, look at the changed parameter (it will now be back at the default, as it wasn't read from the eeprom.bin)

[example_defaults.txt](https://github.com/ArduPilot/ardupilot/files/4295833/example_defaults.txt)

IamPete1, 2021-09-06:
@matthewmcgill I can't reproduce with ` ./Tools/autotest/sim_vehicle.py -v Rover --add-param-file example_defaults.txt`  any reason not to be using `sim_vehicle.py` ?

matthewmcgill, 2021-09-07:
Mostly I didn't use sim_vehicle.py as for my simulations I often run multiple SITL instances with different frames and default parameter files.

IIRC: The issue was that when parameters were changed they were written to the EEPROM file, but because the FORMAT_VERSION didn't change from the default it wasn't written.  When subsequent SITL were started they checked the FORMAT_VERSION in the file, didn't find it and then rejected the file completely.