ArduPilot/ardupilot #13738: Plane Hand throw takeoff  failure in TAKEOFF mode

cesforchina, 2020-03-05:
## Bug report

**Issue details**
[image]
[image]

1.  The plane was thrown by hand on a hillside。As shown in the blue curve above，when the altitude is falling, the mode is in TAKEOFF mode, the throttle is maintained at 2000, and the expected elevation angle is negative.
*  FBW_MIN 16m/s
* FBW_MAX 18m/s
2.  As shown in the airspeed curve above, is it because the airspeed did not reach the value of _FBW_MIN that caused the aircraft to dive to obtain airspeed?
3.  What is the logic of the flight control in this flight mode? The aircraft did not reach the predetermined altitude, and the airspeed was lower than the value of FBW_MIN. Will the aircraft dive? If so, wouldn't the speed not meet the requirements when taking off mean a certain dive into the plane?

**Version**
AP4.0.3

**Platform**
Plane

**Airframe type**
Plane 
RC_out:
ch1,ch2 Aileron
ch3,ch4 Throttle
ch5,ch6 Elevator
ch7        Ruder

**Hardware type**
F4BY

**Logs**
[00000007.zip](https://github.com/ArduPilot/ardupilot/files/4293500/00000007.zip)

cesforchina, 2020-03-06:
> My guess is that this is a support items at this point and should go into the [Plane-4.0 discuss category](https://discuss.ardupilot.org/c/arduplane/plane-4-0/153) until it's confirmed to be a bug.

Ok
I will try to describe my problem in discuss
If this issue is not posted here, I will close it

closed 2020-03-06 by cesforchina

jmachuca77, 2020-03-06:
Probably the altitude is set to absolute instead of relative on the mission 

Best regards,

Jaime Machuca
CEO | Element Aero | www.element.aero
Mail. jaime@element.aero

US Mobile. +1 (720) 610 4585

MX Cel. +52 1 (33) 3945 3350

Este correo electrónico y cualquier archivo transmitido en él, son confidenciales y para uso exclusivo de los individuos y entidades a quienes está dirigido. Si usted no es el destinatario previsto o la persona encargada de recibirlo, y tiene por error este mensaje, queda prohibido y sin validez el uso, difusión, re-envío, reimpresión o copia. Toda oferta y/o aceptación de propuestas comerciales, celebración de contratos u otros actos tendientes a la adquisición de bienes o servicios, así como el establecimiento de cualquier clase de obligación legal para Element Aero SAPI de CV, deberá confirmarse por escrito firmado autógrafamente por funcionario competente, excepto que se cuente con un contrato vigente que autorice el uso de este medio para tales fines. Si usted recibió este correo por equivocación, favor de notificar inmediatamente por este medio a su remitente, y después borrarlo de su correo

cesforchina, 2020-03-06:
> 在执行任务时，高度可能设置为绝对高度，而不是相对高度。元素航空| [www.element.aero](http://www.element.aero) 邮件。jaime@element.aero美国移动。+1（720）610 4585 MX Cel。+52 1（33）3945 3350Estécorreoelectricrónicoy cualquier archivo Transmissionido enél，son confidenciales y para uso exclusivo de los individuos y entidades a quienesestádirigido。没有正确的预装任何人的合法性，请以最正确的方式使用您的个人信息，并在其他情况下禁止使用任何有效的证据。商业合同证明书，公共服务合同证明书和法定服务证明书，以及合法的法人证明书，确认了航空航天公司的合法权益主管人员，除违反常规合同外，还应处以罚款。
> […](#)
> 2020年3月5日下午5:59，cesforchina ***@***.***>写道：我的猜测是，这是目前的支持项目，应纳入Plane-4.0讨论类别直到确认是错误为止。好吧，我将在讨论中尝试描述我的问题。如果此问题未在此处发布，我将其关闭—由于您已订阅此线程，因此您将收到此消息。直接回复此电子邮件，在GitHub上查看或取消订阅。

*  https://ardupilot.org/plane/docs/takeoff-mode.html
*  [image]
[image]
*  The absolute altitude of the take-off point is indeed above 500m, but the effective values of TKOFF_ALT and TKOFF_LVL_ALT shown in the figure above are below 200m，Can it be considered a relative altitude? In other words, the flight controller does not expect to dive due to the absolute altitude problem, unless TKOFF_ALT and TKOFF_LVL_ALT are logically characterized as absolute altitude