ArduPilot/ardupilot #13739: Use of "MOT_SLEW_UP_TIME" leads to bad attitude and crash.

AlvaroRAS, 2020-03-05:
Use of "MOT_SLEW_UP_TIME" leads to bad attitude and crash.

I was testing this parameter on one of our testing copters, and crash; I notice that it happens when mix pitch and roll at the same time, there is a video of the comparison.

Version
Copter 4.0.3 (ChibiOS).

Platform
[ ] All
[ ] AntennaTracker
[X] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Airframe type
Quadcopter

Log
[MSU Tests.zip](https://github.com/ArduPilot/ardupilot/files/4294205/MSU.Tests.zip)

Video
https://www.youtube.com/watch?v=Mq-8wJO7G1M

IamPete1, 2020-03-07:
I think this is expected, your telling it that it cant change the throttle on the motors so quickly. You should re-tune if you increase it. Its meant for stopping esc de-sync.

Pedals2Paddles, 2020-03-07:
Yep. This  is not a feature that you would utilize without a very specific use-case requirement.  And the copter needs to be tuned with the slew rate.  Turning the slew on and off without tuning will have undesired results.  And using the slew without some actual thought out purpose will probably also not work out well.

If you simply set this parameter for some random number and went flying just to see what happened, then this is pretty much what I would expect t happen. Did you have a use case you're working towards? Or was this just a random test of a parameter?

closed 2021-08-08 by IamPete1