ArduPilot/ardupilot #13757: Loop-rate scheduled tasks always slip at 800Hz

andyp1per, 2020-03-08:
I added a dummy callback to the runcam driver (doesn't matter where it is this was just convenient):

```
void AP_RunCam::dummy() {
}
```
I then included this in the vehicle schedule table at the loop rate:
```
    SCHED_TASK_CLASS(AP_RunCam,    &vehicle.runcam,         dummy,                   LOOP_RATE, 50),
```
I then set ```SCHED_DEBUG = 2``` and ```SCHED_LOOP_RATE = 800```

The empty task is then reported to always slip:

```
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
Scheduler slip task[51-dummy] (2/1/50)
```

My suspicion is that this is because 800Hz is close to the backend IMU rate of 1000Hz. 800Hz gives a tick length of 1250us and the IMU period is 1000us. Since we always wait for an IMU sample before scheduling the next tick if we overrun by 250us I think we will always slip loop rate tasks. Just a theory.

@tridge @peterbarker I could use some help figuring this out

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

andyp1per, 2020-03-08:
Effectively fixed by https://github.com/ArduPilot/ardupilot/pull/13758

IamPete1, 2021-08-16:
fixed by https://github.com/ArduPilot/ardupilot/pull/13741

closed 2021-08-16 by IamPete1