ArduPilot/ardupilot #13767: Antenna tracker: calibration mode instead of mounting a flight controller

LimitlessGreen, 2020-03-09:
## Feature request

**Problem**
_I see a lot of people specially in the HD FPV community who avoid ArduPilot AntennaTracker, because they like to build smaller tracker setups. They basically request 2 Features:_
- [ ] A solution, without mounting the flight controller on the tracker's head
- [ ] Using stepper motors instead of servos

**Solution**
_Instead of mounting the controller, the pilot can point the antenna before start manually to the vehicle._

**Platform**
[   ] All
[x] AntennaTracker
[   ] Copter
[   ] Plane
[   ] Rover
[   ] Submarine

**Motivation**
I saw multiple issues about stepper motors (#124, #9172 or #10030), but they didn't got much attention.
Now, a guy named Miroslav Kolinsky build a beautiful 3D-printed tracker design. He will open source it in the future, but unfortunately the tracker software in the current state isn't suitable for it.
[image]
I'm sure there are many people here, who want to build it on their own.

IamPete1, 2020-03-09:
The idea is to use steppers so you can move a known amount from the starting position? This is a move from closed loop feedback to open loop. I don't really see any reason why mounting the flight controller to the head is a big disadvantage, ArduPilot supports plenty of small flight controllers.

knputyi, 2020-03-09:
subbed to the idea

LimitlessGreen, 2020-03-09:
That's the idea.

I'm thinking in terms like "start with as little hardware as possible".
If breaking the things down, only the trackers position (maybe could also approximately be the start position of the vehicle) and it's antenna direction on start is needed to calculate the movements based on telemetry during the flight. (For sure you have to know the step size of the motors)

I thing the main reasons are: cable management, small/cheap setups

LimitlessGreen, 2020-06-30:
What is, if we drop the open loop stuff and focus on stepper motor support?
I'm thinking about Trinamic drivers with SPI and/or UART.