ArduPilot/ardupilot #13777: Updating sectors from OBSTACLE_DISTANCE was shifted 45 degree counter clockwise

bigboy061293, 2020-03-10:
**Issue details**

To my understanding, by sending the array of 72 elements via OBSTACLE_DISTANCE message will automatically update the proximity information which could also be set up by DISTANCE_SENSOR message. However, when I sent 72 elements, from 0, 100, 200... 7100; offset was -55.08 degree; incremental degree was 1.53 degree; I expected:
. The smallest element in sector 7 would be min_distance (this case is 100)
. The smallest element in sector 0 would be 2200, because 21st element is in the region -22.5 and +22.5
. The smallest element in sector 0 would be 5100, because 52nd element us in the region of +22.5 and 67.5

But here is what I get:

[image]

Similarly, If I don't do the offset -55.08 degree, I got:

[image]

It should be 100 in sector 0, 1500 in sector 1 and 4400 in sector 2

Anyone please have a look on this?
Thanks in advance.

**Version**
ArduCopter 4.0

**Platform**
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_Frame X
**Hardware type**
_Cube

**Logs**
_Please provide a link to any relevant logs that show the issue_

bigboy061293, 2020-03-10:
For more detail on this problem, below are the photo:

[image]

The left window displayed actual objects appearing in front my radar.
In MP, the indicating red circles also interpreted the same
But proximity sector did not!

IamPete1, 2021-09-12:
@bigboy061293 I wonder if your sensor is mounted upside down? We have a orientation parameter for that. 

https://ardupilot.org/copter/docs/parameters.html#prx-orient-proximity-sensor-orientation

magicrub, 2021-09-12:
Please include a param file or a log (which we can use to extract the params). I have a feeling there is something funny in your configuration.

bigboy061293, 2021-09-12:
> @bigboy061293 I wonder if your sensor is mounted upside down? We have a orientation parameter for that.
> 
> https://ardupilot.org/copter/docs/parameters.html#prx-orient-proximity-sensor-orientation

I am not so sure and did not check it at the time I test. As I remember, this parameter was not touched.
This PRX sensor is fed to Arducopter via mavlink by Obstacle Distance message!

magicrub, 2021-09-12:
Please also include an example mavlink msg. Have you tried packets smaller than 72?

bigboy061293, 2021-09-12:
> Please also include an example mavlink msg. Have you tried packets smaller than 72?

I guess I had done it already. I was stuck at that time and no chances to test again in newer copter version.

You can refer to it here. It's nothing but just sending a normal mavlink message using mavutil https://github.com/bigboy061293/radarCAN/blob/80ffde321862ab5a90fa1239386339b8c663e641/on650/sendToAPBK1503.py#L439

rmackay9, 2021-09-13:
If the mavlink message is being sent from ROS then I think the issue is actually in mavros.