ArduPilot/ardupilot #13778: MAV_CMD_CONDITION_GATE Support

Saijin-Naib, 2020-03-10:
## Feature request

https://github.com/mavlink/qgroundcontrol/pull/8536

> Problem:
> If you start/stop camera triggering in each transect at the survey edges you do not get precise coverage of the survey area. This is due to waypoint acceptance triggering prior to the vehicle actually getting to the wp position. So for survey edge entry you tend to get images prior to the survey edge. And for survey edge exit you will miss images which go all the way out to the edge.
> 
> The new CONDITION_GATE command works just like a wp from a standpoint of the vehicle flying to it. But it delays the mission state machine until the vehicle flies through an imaginary line which is orthogonal to it and the previous waypoint. If you put this in front of a camera trigger then it means the triggering will occur at the gate position instead of before it. Hence the triggering will start/stop right at the survey edges.
> 
> The goal is to no longer need to capture images throughout the survey including the turnarounds to get correct coverage of the survey area.
> 
> Note: This requires firmware support for MAV_CMD_CONDITION_GATE which is currently a WIP.

https://discuss.ardupilot.org/t/3dr-solo-compatibility/53216/15?u=saijin_naib

> Also just to continue it’s likely the firmware differences for generating patterns is going to diverge more. For example I am currently working on support for MAV_CMD_CONDITION_GATE. This allows for much more precise triggering of starting/stopping photo triggering right at the edges of the survey area. Using this allows you to not need need to specify “Images In Turnarounds” to make sure you are getting good coverage at the edges of your survy polygon.

DonLakeFlyer, 2020-03-11:
The other thing missing is that ArduPilot does not support MAV_CMD_DO_SET_CAM_TRIGG_DIST.param2=1 to trigger immediately

Saijin-Naib, 2020-03-11:
@DonLakeFlyer ,

Is that different than what I asked for here:  
https://github.com/ArduPilot/ardupilot/issues/13779

And what was created by @Gone4Dirt  here:  
https://github.com/ArduPilot/ardupilot/pull/13791

DonLakeFlyer, 2020-03-11:
Oh, sorry I didn't see that one.

Saijin-Naib, 2020-03-11:
No, not a worry at all, just making sure that the way I asked for the parameter was not different than how you cited it above (`MAV_CMD_DO_SET_CAM_TRIGG_DIST.trigger=1` vs `MAV_CMD_DO_SET_CAM_TRIGG_DIST.param2=1`)

DonLakeFlyer, 2020-03-11:
I had a typo in my comment above. trigger is on param3.