ArduPilot/ardupilot #13779: MAV_CMD_DO_SET_CAM_TRIGG_DIST.trigger=1 Support

Saijin-Naib, 2020-03-10:
https://discuss.ardupilot.org/t/3dr-solo-compatibility/53216/15?u=saijin_naib

> For example support for MAV_CMD_DO_SET_CAM_TRIGG_DIST.trigger=1, which allows you to start distance based triggering with an initial image triggered right away. This allows you to catch the image right at survey edge entry.

DonLakeFlyer, 2020-03-11:
The trigger is on param3

Gone4Dirt, 2020-03-11:
I have used param3 in the pr.  The packet was already definied in the mavlink message.  All I have done is added support for it in ArduPilot

DonLakeFlyer, 2020-03-11:
Yup, thanks for adding this. So far test results with QGC and other firmwares are showing the combination of these two are working to provide higher quality survey coverage without needing to over cover the area.

Saijin-Naib, 2020-03-11:
> Yup, thanks for adding this. So far test results with QGC and other firmwares are showing the combination of these two are working to provide higher quality survey coverage without needing to over cover the area.

This is so crucial, especially in suburban environments when there are lots of concerns over privacy, access, safety, etc. Not having to grossly overfly is so wonderful.

closed 2020-03-30 by tridge