ArduPilot/ardupilot #13790: AP_RangeFinder: Driver Request for Benewake new TF-Luna low-cost distance sensor

bys1123, 2020-03-11:
Benewake released their new distance sensor TF-Luna series SJ-GU-TF-Luna A01, target drone market.
Datasheet: https://1drv.ms/b/s!AmscxcpJYtuYmTcSOcB6W16AOyF0
https://www.compotek.de/news/tf-luna-lidar-module-fuer-erkennungsapplikationen/

bkzspam, 2020-04-28:
@rmackay9 Hi, Is any update on this?

mboland, 2020-06-12:
+1 for an update as I have purchased one and landed here looking for the reason it won’t be seen on serial.

andyp1per, 2020-06-12:
I have on of these and I believe it is working fine?

peterbarker, 2020-11-25:
We have support for this in master.

closed 2020-11-25 by peterbarker

bys1123, 2020-11-25:
@peterbarker Thank you! Is there any commits or PR related this driver?

bys1123, 2020-11-25:
Ha, TF-luna changed the protocol to compatible with old driver.