ArduPilot/ardupilot #13810: Very few dataflash log-items of type POS,EKF2,RCIN,PM  (most likely others too)

AndKe, 2020-03-14:
## Bug report

**Issue details**
The flight logs  have very few lines of type POS,EKF2,RCIN,PM

**Version**
4.0.1 and 4.0.3

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad (Solo)

**Hardware type**
Solo "Green cube"

**Logs**
https://drive.google.com/open?id=19dEsSnAozU9Q3mrDnRE_UuOKJIUOVDDa
https://drive.google.com/open?id=1oPNQQRvwhSNNjfALZTokwMZL9ahpvbSA

AndKe, 2020-03-14:
LOG_BITMASK =176126d  (nothing unusual)
@Pedals2Paddles  - what do you think, can you confirm?

Pedals2Paddles, 2020-03-15:
176126 is the default and should be fine. The log does indeed seem to be missing a lot of data that shouldn't be missing. The Performance Monitor log shows the loop time is pretty long.

rmackay9, 2020-03-16:
Re the long loop times, @tridge is working on some performance improvements because we seem to have a couple of performance regressions that have crept in between 3.6 and 4.0.  I wonder if it might be useful to go back to Copter-3.6 and see if the logging performance improves.  This would highlight if it's a regression or not and we tend to prioritise regressions.

AndKe, 2020-03-16:
oh yes it is a regression, please see https://drive.google.com/open?id=11fb4funMyXcx09i_EwDOJliaJUaK6ZlM
3.7.0-dev , behaving fine.

AndKe, 2020-03-16:
3.6.1-rc1 : https://drive.google.com/open?id=17YRt61MUN10JyJ_USQqjdoNevcwJck6w  
is normal

rmackay9, 2020-03-17:
@AndKe, OK thanks, I've added it to our [Copter-4.0 issues list](https://github.com/ArduPilot/ardupilot/issues/12547).

AndKe, 2020-03-19:
This is related to "Green Cube" build.
I fresh installed OpenSolo4 + Stock cube 4.0.1 for one Solo   , and same , but "Green Cube 4.0.1" build on another.  (Both via Solex)
The stock Cube2.0 performs well, while the Green cube, (Basically Cube 2.1+solder jumper) has this issue.
Cube 2.1 with AC 4.0.1 in X8 configuration  run fine too.
So what is wrong with Green Cube Solo build ? @Pedals2Paddles - maybe you know the difference?

AndKe, 2020-03-19:
Correction: It is not the green cube build, but the green cube itself(?)
I used 4.0.1 stock cube firmware on it, and too show these symptoms of some parameters being logged slow, and maybe performance problems.
[image]

AndKe, 2020-03-19:
One could say it is a performance issue, but how can it be that it affects only certain log items ?
POS & EKF2 logs have a very slow rate, while GPS is logged flawlessly.
- Strange!

AndKe, 2020-03-19:
Breakthrough!
This affects only IMX6 logging, NOT internal SD card logging.
In other words: SCP downloded logs are missing much data, while mavlink-downloaded logs are complete.
Attaching screenshot of the same log/log items, on top you can see SSH IMX6 log, below is the same log, MAVLink downloaded from the uSD:

[image]

So it it IMX6's logging that fails, maybe going bananas over some new log items in AC4.x ?

AndKe, 2022-02-13:
Closing, because this issue should be solved on the IMX6 side, rather than here.

closed 2022-02-13 by AndKe