ArduPilot/ardupilot #13811: Drone crash after conducting Flip mode

KimHyungSub, 2020-03-14:
## Bug report

**Issue details**

I tested the Flip mode on SITL and the Gazebo simulator with SITL.
Firstly, when I tested the Flip mode on the Gazebo, the drone was crashed after finishing the Flip mode. You can see the demonstration video in the below link.
(https://youtu.be/kxjA4pd9Y8k)

Secondly, when I tested the Flip mode on the SITL, the drone was not immediately crashed when I triggered the Flip mode. But, it was also crashed after I send several Flip mode commands. You can also check the demonstration video in the below link.
(https://youtu.be/jrq-jnod-Sc)

Also, I confirmed that the altitude of the drone is decreased after the drone conducted the Flip mode. I am not sure that this is normal operation or not because I never try to execute the Flip mode in real environment. But, if decreasing the altitude is nature behavior of the Flip mode, it seems likely to me that it needs to check current altitude of the drone to safely conduct the Flip mode. Also, the documentation (https://ardupilot.org/copter/docs/flip-mode.html) mentions that "Give yourself at least 10m of altitude before trying flip for the first time!". But, I cannot find any code to check whether the current altitude of the drone is more than or equal to 10m.

**Version**
- Ubuntu 16.04
- Copter-4.0.0
- gazebo7 libgazebo7-dev

**Platform**
[  ] All
[  ] AntennaTracker
[V] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
ArduCopter

**Hardware type**
SITL and Gazebo simulator with SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

khancyr, 2020-03-15:
I am not surprised, the flip was always losing some altitude.

KimHyungSub, 2020-03-15:
@khancyr Do you think that checking the altitude is not required?

rmackay9, 2020-03-16:
There's certainly room for improvement in our flip code but I think the way it's written at the moment it will always lose altitude.  I've changed the title of this request and turned it into an enhancement request.

KimHyungSub, 2020-03-16:
@rmackay9 Can we simply check a minimum required altitude (such as 10m) before the drone conducts the flip mode? or there is also another way. When the drone conducts the flip mode, we can measure how much altitude lose (i.e., X meters). After that, when the drone re-execute the flip mode, we can check the current altitude is less than or equal to X meters. These methods are kind of naive. But, I hope that this will be a starting point to improve the flip mode :)

rmackay9, 2020-03-16:
@KimHyungSub,

It's certainly possible although we would probably need to add a FLIP_ALT_MIN parameter so that users can disable the altitude check.  The check should also be using the best estimate of our altitude above terrain (there's a method somewhere in Copter that returns this) because we can't always assume the earth is flat.

I'm afraid I can't personally take on this enhancement request though, perhaps another dev can.

kd0aij, 2020-03-20:
This looks like another good place for scripting