ArduPilot/ardupilot #13815: Checking min/max angle of gimbal

KimHyungSub, 2020-03-15:
## Bug report

**Issue details**
(1) The current master version does not check min/max angular position of mount such as a gimbal when the drone receives 'Do-Mount-Control' mission.

I included 'Do-Mount-Control' which contains 190 (roll), 100 (pitch), 190 (yaw) in my mission.
Also, I used in the below parameter setting.
MNT_ANGMAX_ROL = 4500 (centi-degrees)
MNT_ANGMAX_PAN = 4500 (centi-degrees)
MNT_ANGMAX_TIL = 4500 (centi-degrees)

But, when the drone received the 'Do-Mount-Control' command, it showed in the below result.
MOUNT_STATUS
pointing_a = 9999 (pitch)
pointing_b = 19000 (roll)
pointing_c = 19000 (yaw)

(2) It seems likely to me that ArduPilot needs min/max check according to the parameters such as MNT_ANGMAX_ROL.
When I included 'Do-Mount-Control' which contains 1900 (roll), 1000 (pitch), 1900 (yaw) in my mission, it showed in the below result. But, some gimbals physically cannot rotate the big digresses such as 1900.
[image]

**Version**
ArduPilot master

**Platform**
[V] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
I used quad-copter.

**Hardware type**
I used SITL.

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-03-16:
I've looked at this a bit and I didn't fix it because I found that the MNT_ANGMAX_ROL/PAN/TIL angles are "body frame" limits while the angles provided in the do-mount-control messages are "earth frame".

The issue is most obvious with yaw (aka "pan").  A do-mount-control might provide a target yaw of 0 degrees (meaning point North).  The mount can easily accept that if the vehicle is also pointing north but it can't if it's pointing south.

I think it's possible to apply these limits but it needs to be done at the individual mount driver level after the conversion from earth-frame to body-frame has been done.  It's already been done for the servo mount driver.

KimHyungSub, 2020-03-16:
@rmackay9 I was kind of confused whether MNT_ANGMAX_ROL/PAN/TIL angles are earth frame or body frame. Can we clearly mention these things on ArduPilot's documentation? (https://ardupilot.org/copter/docs/parameters.html)

IamPete1, 2022-09-11:
@rmackay9 I wonder if this has been resolved by your recent improvements?

rmackay9, 2022-09-11:
This problem still persists sadly. There's a duplicate issue for it here so I suppose we could close one of them https://github.com/ArduPilot/ardupilot/issues/21547

rmackay9, 2023-06-08:
I think this problem is fixed by PR https://github.com/ArduPilot/ardupilot/issues/21547 which I suspect will be merged by next Tuesday.  If someone wants to test it before then I'm happy to create a binary just tell me which autopilot you're using. txs!

closed 2023-06-13 by tridge