ArduPilot/ardupilot #13819: Cube Purple doesn't probe compass from MPU9250

MallikarjunSE, 2020-03-16:
## Bug report

**Issue details**
The “COMPASS AK8963:probe_mpu9250” line is missing from the hwdef.dat file of Cube Purple and It doesn't probe compass.

_Please describe the problem_
Looks like a small line is missing from hwdef.dat file which lets AK8963 to be recognised on Cube purple boards. I tried adding the line to latest master and commenting out the ICM20948 compass probe. It works. Hoping to see this in next beta or stable. 
https://discuss.cubepilot.org/t/internal-compass-not-recognised-on-cube-purple/2953/5
**Version**
_What version was the issue encountered with_ Doesn't work on any for the stable, beta, or master. 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe 
Quadcopter, Fixed wing

**Hardware type**
_What autopilot hardware was used? 
Cube Purple

**Logs**
_Please provide a link to any relevant logs that show the issue_
[](url)

MallikarjunSE, 2020-03-17:
https://github.com/ArduPilot/ardupilot/pull/13825

IamPete1, 2021-08-08:
Since it was decided to close the PR, looks like there is some sample rate issue with the compass, I'm also going to close this issue.

closed 2021-08-08 by IamPete1