ArduPilot/ardupilot #13855: ArduPlane / QuadPlane - SITL Failure.

beachbum1973, 2020-03-21:
## Bug report

**Issue details**

Executing SITL via Mission Planner for "ArduPlane" & "QuadPlane" results in failure of SITL to run and mission planner to connect to SITL (connection timeout). Selected model from pull-down, either "plane" or "quadplane"

ArduCopter SITL is working correctly.

Have updated Mission Planner, deleted folders (default_params and plane SITL folders) to ensure downloading of latest SITL.

Have also recently used this quadplane SITL on this PC and was working.

SITL window - highlighted a couple of lines, not sure if these are the problem(s):

_Set parameter SIM_SPEEDUP to 1.000000
Home: -34.100000 138.100000 alt=6.000000m hdg=0.000000
Starting sketch 'ArduPlane'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
**setup:329: bad header in setup - erasing**
Loaded defaults from C:\Users\xxxxxxx\Documents\Mission Planner\sitl\default_params/plane.parm
Smoothing reset at 0.001
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
validate_structures:419: Validating structures
Loaded defaults from C:\Users\xxxxxxxx\Documents\Mission Planner\sitl\default_params/plane.parm
**Short write on UART: 0 < 6**
Running: sh dumpstack.sh 3652 >dumpstack_ArduPlane.3652.out 2>&1
Failed_

**Version**
Mission Planner 1.3.70 build 1.3.7277.34800

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
Win10

**Logs**
Not sure if detailed log exists.. unable to easily find a log file.
Does a detailed log exist? if so, where?

alejandrissimo, 2020-03-25:
Same problem here, No plane simulation on any of my computers.

peterbarker, 2020-03-25:
OK, so this sanity check is designed to stop us emitting corrupt mavlink (or other) packets.  See, we're supposed to check there's free space in a uart's (think "serial port" if that was beyond you...) buffer before we stuff a message into it.  If we ever write() more bytes into a uart than we have space to store then we throw this error - and it has caught bugs.

So.   We need to work out who is writing this and why.

Sadly, being a Windows machine, you can see our efforts to provide a stacktrace are failing woefully.  While it is possible to get a stacktrace under Windows, it isn't something we've implemented.

6 bytes isn't a lot.  Not going to be a mavlink packet.  Probably somebody doing a raw print somewhere....

The fact that this is reliable leads me to think that it isn't a thread race condition (and we don't have too many of those in SITL anyway).

@meee1 - I don't suppose you could come up with a pair of git hashes where the problem might have come up?  I'm guessing newer versions of MP might mean newer SITL binaries?

Can anyone confirm this is only a problem for QuadPlane, not for any other model?  @alejandrissimo - were you trying to simulate a QuadPlane or just a Plane?

alejandrissimo, 2020-03-26:
Hi. This is happening in different version s of mission planner just with
plane models. Quad plane, plane, etc. On the other hand rover, heli and
quad is working well.

I tried on different computers and the problem is the same. Also same error
message on the console.

peterbarker, 2020-03-26:
I've created a PR to resolve this.

closed 2020-03-26 by tridge

jankrystianserafim, 2023-05-10:
Just pubbing this issue - I am having the same problem still

PR #13892 (2020-03-26): Plane: check channel space before sending servo output raw in SITL CPU failsafe
.... why this wasn't a problem on Linux I don't know.

An interesting difference in behaviour between Linux and Windows SITL to be remembered.

Closes #13855