ArduPilot/ardupilot #13860: MAVLink 2.0 message signing can be broken under GPS spoofing

KimHyungSub, 2020-03-22:
## Bug report

**Issue details**
As you already know, MAVLink 2.0 supports the message signing to prevent drones from message spoofing/replay attacks.
However, I noticed that the message signing can be broken under GPS spoofing.
The timestamp is key to prevent the replay attack as shown in the below.
[image]

But, when I conducted GPS spoofing, ArduPilot updated both system clock and timestamp for the message signing. You can see a demonstration video. (https://youtu.be/JsUApnQAe7o)
It means that anyone controls the timestamp and then conducts some replay attacks without the secret key.

**Version**
3DR IRIS+, Pixhawk1, Copter 3.5.5 (I used Copter 3.5.5, but I sure that the attack is available on the master version of Copter).

**Platform**
[ V] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
All

**Logs**
_Please provide a link to any relevant logs that show the issue_

peterbarker, 2020-03-22:
On Sun, 22 Mar 2020, Hyungsub wrote:

> As you already know, MAVLink 2.0 supports the message signing to prevent
> drones from message spoofing/replay attacks.
> However, I noticed that the message signing can be broken under GPS
> spoofing.
> The timestamp is key to prevent the replay attack as shown in the below.
> Picture1
> 
> But, when I conducted GPS spoofing, ArduPilot updated both system clock and
> timestamp for the message signing. You can see a demonstration video.
> (https://youtu.be/JsUApnQAe7o)
> It means that anyone controls the timestamp and then conducts some replay
> attacks without the secret key.

OK, that sounds very bad.

I'm a little unclear as to what your video is showing, however.  It 
doesn't help that my playbak here has the first half of the video rather 
blurry :-(

What I seem to be seeing there is the signing timestamp moving into the 
future - not the past.  That doesn't allow for a replay attack.

We do have protection for the timestamp going backwards during a session - 
the method "update_signing_timestamp" contains that.

We also persist the signing timestamp in storage for future boots - and 
there's an additional check in there that we don't set the persisted key 
backwards - that's in save_signing_timestamp()

Would be great to be able to reproduce this attack.  You're saying that 
you're able to reset the signing timestamp to a time in the past - that's 
definitely a bad thing and something we should investigate.  It's also 
getting around the code we have in place to disallow just that.

It would be good if you could replicate on master - or at least a
current stable firmware.

tridge, 2020-03-22:
This doesn't really make sense. We force the 64 bit timestamp to only ever go forward.

tridge, 2020-03-22:
A few notes:
 - as I pointed out, we force the timestamp to only go forward
 - we only use the GPS timestamp once while booting, from then on we use the system clock as offset from that initial GPS timestamp
So I don't think your described issue makes much sense. 
I also suggest you investigate issues using SITL. In SITL you have complete control over GPS (see libraries/AP_HAL_SITL/sitl_gps.cpp) so much easier to look for issues.

KimHyungSub, 2020-03-22:
@peterbarker @tridge Let me clarify the issue.
"The timestamp is only go forward" is true. But, ArduPilot start its timestamp from 1 January 1970 before ArduPilot detects GPS signals. In this situation, I can set the timestamp such as it came from 1 January 2020 by using the GPS spoofing.

Detail steps.
Let's assume that current date is 22 March 2020 and attackers capture MAVLink messages during 1 January 2020.
1) Attackers spoofs GPS signals which contain time info. such as 1 January 2020.
2) A user turns on the drone and set up message signing.
3) ArduPilot uses the timestamp 1 January 2020.
4) The attackers replay the captured MAVLink message.

peterbarker, 2020-03-23:
On Sun, 22 Mar 2020, Hyungsub wrote:

> @peterbarker @tridge Let me clarify the issue.
> "The timestamp is only go forward" is true. But, ArduPilot start its timestamp from 1 January 1970 before ArduPilot detects GPS signals. In this [...]

OK, this it should not do.  We should have stored the 22 March 2020 in 
persistent storage, loaded it when the autopilot booted, and rejected the 
1 January 2020 timestamp as being in the past.

Can you confirm that that is not the case - that we are NOT persisting the 
higher timestamp?

If you can supply a git hash and a platform then we can perhaps get you a 
firmware which adds debug around the persistent to see if we can find the 
problem!

Note that we are aware of a different DOS issue around boot-time.  But 
that would be unrelated to what you're apparently seeing.

>  4. The attackers replay the captured MAVLink message.

KimHyungSub, 2020-03-23:
@peterbarker thanks for your answer. The timestamps on the persistent storage are available even if the user flush and then update the storage?

tridge, 2020-03-23:
The only scenario would be if the user deliberately reset all of storage. The attacker has to trick the victim into doing something to reset storage. How does the attacker do that? Even upgrading firmware won't reset the signing timestamp.
It is stored in hal.storage, which is not on the microSD card. It is stored along with parameters

peterbarker, 2020-03-25:
@KimHyungSub are you satisfied that the conditions under which spoofing is possible are limited to where an autopilot's storage has been entirely reset - so there's no pre-existing timestamp for us to initialise from - and that the spoofing would have to occur before legitimate GPS with current timestamps has been seen?

KimHyungSub, 2020-03-25:
@tridge @peterbarker 
If we only consider the replay attack what I described, I understand the attack surface is narrow. But, there are many possible attack scenarios because anyone can compromise the timestamp. Also, the root problem is that the secret key and timestamp are exposed to some attacks because there is no any secure secret key exchange algorithm as well as attackers still make the timestamp going to forward. FYI, based on my experiments, it's not difficult for attackers to make the spoofing would have to occur before legitimate GPS with current timestamps has been seen.

peterbarker, 2020-03-25:
> @tridge @peterbarker
> If we only consider the replay attack what I described, I understand the attack surface is narrow. But, there are many possible attack scenarios because anyone can compromise the timestamp. 

Sorry, we're still having trouble understanding the proposed attacks.  It would be great if you could create an entirely-SITL-based attack example we could play with ourselves.  Sadly, I don't have an SDR setup to play with here :-)

> Also, the root problem is that the secret key and timestamp are exposed to some attacks because there is no any secure secret key exchange algorithm as well as attackers still make the timestamp 

Well, it's a secret key which should only be set up once-only via a directly-connected USB computer.  But yes, that should be documented as part of any setup procedure; do you have any suggestions on where we should mention that fact?

> going to forward. FYI, based on my experiments, it's not difficult for attackers to make the spoofing would have to occur before legitimate GPS with current timestamps has been seen.

So just to be clear.... the ONLY time this can be attacked is if storage has just been reset?

If a user could replay packets any time the vehicle hasn't seen any (legitimate) satellites, we might be panicking - please say so if that is the case!

KimHyungSub, 2020-03-25:
@peterbarker 
- Give me some time to conduct the experiment. 
- It seems likely to me that using the USB channel is just recommendation. Users can still use any communication channel. Am I right?
- Yes, the current  scenario is only possible when the user reset the storage.
- Based on my experiments, when the drone receives both legitimate and spoofed satellites info, the  UBlox GPS receivers trusted spoofed satellites because its signal strength and the number of satellites were more than legitimate's one.

peterbarker, 2020-03-26:
>     * Give me some time to conduct the experiment.

Thanks!

>     * It seems likely to me that using the USB channel is just recommendation. Users can still use any communication channel. Am I right?

Well, our recommendation would be along the lines of "ensure your communication channel used to set the secret key is secure itself" :-)  USB is one example, but on Linux you could have a secure ethernet connection or whatever.  Basically, we have to document ourselves out of this problem by adding a warning whereever it's required.

>     * Yes, the current  scenario is only possible when the user reset the storage.

Thanks for confirming.  We should also document that somewhere - have you suggestions where we should?  Note that setting up the signing key in the first place should also be setting an 'initial timestamp'.

>     * Based on my experiments, when the drone receives both legitimate and spoofed satellites info. at the same time, the  UBlox GPS receivers trusted spoofed satellites because its signal strength and the number of satellites were more than legitimate's one.

It would be quite interesting to see what the UBX messages we're currently logging are saying about all of that.  Can you detect the spoofing is going on from any of the metadata messages we're getting from the receiver?

Also, as far as we're aware, nobody has shown a GPS attach which allows a vehicle to be manipulated.  Jamming, sure - but not deliberately manoeuvring the vehicle, for example.  Do you know otherwise?

KimHyungSub, 2020-03-26:
@peterbarker 
- I my opinion, you guys already have a wiki webpage to explain how to set up packet signing. In order to learn how to use packet signing, users have to review this webpage. Could we insert the warring message in here (https://ardupilot.github.io/MAVProxy/html/modules/signing.html)? 

- Yes, we can detect the spoofing even though the UBlox cannot detect all cases. When the UBlox detects any spoofing attempt, it sends warning messages to ArduPilot. I posted the Ublox GPS spoofing indicator in here (https://github.com/ArduPilot/ardupilot/issues/11239).

- Actual GPS attack case: I know one specific case even though they targeted a smartphone navigation system. (https://www.usenix.org/sites/default/files/conference/protected-files/security18_slides_zeng.pdf). I also mimicked their method.

- Lastly, you mentioned DoS during boot-time. Could you share how GPS can cause DoS during boot-time?

peterbarker, 2020-03-26:
>     * I my opinion, you guys already have a wiki webpage to explain how to set up packet signing. In order to learn how to use packet signing, users have to review this webpage. Could we insert the warring message in here (https://ardupilot.github.io/MAVProxy/html/modules/signing.html)?

Good news is that it's already there: 
[image] :-)

>     * Yes, we can detect the spoofing even though the UBlox cannot detect all cases. When the UBlox detects any spoofing attempt, it sends warning messages to ArduPilot. I posted the Ublox GPS spoofing indicator in here (#11239).

Cool.  Definitely an enhancement.  It might be something that we need to optionally-enable, but I think it'd be nice to have.

>     * Actual GPS attack case: I know one specific case even though they targeted a smartphone navigation system. (https://www.usenix.org/sites/default/files/conference/protected-files/security18_slides_zeng.pdf). I also mimicked their method.

Thanks for the link!  Very interesting.

>     * Lastly, you mentioned DoS during boot-time. Could you share how GPS can cause DoS during boot-time?

After chatting with tridge, I believe I was mistaken in this.  I thought there might be a vulnerability where the timestamp could be advanced.

Thanks for looking into this.  It's always good to have people looking over this sort of code.  Please let us know if you find things we need to address - bit perhaps in a new issue as I believe this one has been resolved.

closed 2020-03-26 by peterbarker