ArduPilot/ardupilot #13864: Validate parameter values

khancyr, 2020-03-23:
## Feature request

**Is your feature request related to a problem? Please describe.**
Currently we only parse the parameter documentation.
We could easily parse and validate that we don't have duplicate value in @value fields

rishabsingh3003, 2020-03-23:
Please assign me to this issue! I'd like to raise a PR :)

rishabsingh3003, 2020-03-23:
@khancyr while fixing this issue I found out there are a number of duplicate values already present in the master. For eg: Copter has the following duplicates:

[image]

Should the PR I push also fix these duplicates? Or should that be a separate PR?

khancyr, 2020-03-23:
in same PR that is better, otherwise you won't pass the test ^^

PR #13867 (2020-03-23): Tools: Validate that there are no duplicate values in @value fields
This PR fixes #13864 

The duplicate values thus found have also been rectified in this PR.
closed 2020-03-23 by peterbarker