ArduPilot/ardupilot #13868: Scripting: incorrect buzzer tones cause reboot

IamPete1, 2020-03-23:
While messing with scripting I have found some buzzer tones will cause the flight controller to reboot. 
notify:play_tune('MB L16 4') for example. I guess the issue is somewhere in the [mml player](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Notify/MMLPlayer.cpp). It seems not to be a watch dog reboot.

khancyr, 2020-03-23:
Can you replicate with SITL ?

IamPete1, 2020-03-23:
No not in SITL, but I can crash a cube orange with that line in the script at all, even commented out @WickedShell

IamPete1, 2020-03-23:
https://youtu.be/806FCNStjEg

IamPete1, 2020-03-23:
Ok, a little more info and some logs, It crashes and watchdog reboots when it hits that line in the script. If the line is right at the start it will crash before the watchdog has been enabled resulting in a permanent lockup.
[Logs.zip](https://github.com/ArduPilot/ardupilot/files/4371265/Logs.zip)

IamPete1, 2020-03-23:
Have done some messing about with different tones, it seems any invalid/unexpected sequence could cause it. I guess there is some bug in the backend somewhere that is letting invalid stuff get down to the Hal level hence the difference in between SITL and 'normal'. Maybe some one linux could test the SITL buzzer.

peterbarker, 2020-03-23:
Is the sequence marked as "repeat"?

IamPete1, 2020-03-23:
not in this instance, but correct repeat and single tones work as expected.

WickedShell, 2020-03-23:
Replicated in SITL, you just need to add `--enable-sfml --enable-sfml-audio` to your configure line, and used the REPL. Looks like an infinite loop (I trimmed all the other repeated frames):

``` gdb
Thread 6 "arduplane" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffef44a700 (LWP 29842)]
HALSITL::ToneAlarm_SF::set_buzzer_tone (this=0x55555584ff71 <HALSITL::Util::_toneAlarm>, frequency=0, volume=0, duration_ms=0)
    at ../../libraries/AP_HAL_SITL/ToneAlarm_SF.cpp:23
23	    if (!AP::sitl()->tonealarm_sim.is_enabled()) {
(gdb) where
#0  HALSITL::ToneAlarm_SF::set_buzzer_tone (this=0x55555584ff71 <HALSITL::Util::_toneAlarm>, frequency=0, volume=0, duration_ms=0)
    at ../../libraries/AP_HAL_SITL/ToneAlarm_SF.cpp:23
#1  0x00005555556b2fa6 in HALSITL::Util::toneAlarm_set_buzzer_tone (this=0x55555584f980 <utilInstance>, frequency=0, volume=0, duration_ms=0)
    at ../../libraries/AP_HAL_SITL/Util.h:51
#2  0x000055555575cb67 in MMLPlayer::stop (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:45
#3  0x000055555575caae in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:26
#4  0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#5  0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#6  0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#7  0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#8  0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#9  0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#10 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#11 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#12 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#13 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#14 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#15 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#16 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#17 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#18 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#19 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#20 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#21 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#22 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#23 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#24 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#25 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#26 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#27 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#28 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#29 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#30 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#31 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#32 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#33 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#34 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#35 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#36 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
[... cut ...]
#42 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#43 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
#44 0x000055555575cf1b in MMLPlayer::next_action (this=0x555555901cd0) at ../../libraries/AP_Notify/MMLPlayer.cpp:136
#45 0x000055555575cb19 in MMLPlayer::play (this=0x555555901cd0, string=0x555555901d08 "MB L16 4") at ../../libraries/AP_Notify/MMLPlayer.cpp:39
```

IamPete1, 2020-03-23:
I have reproduced via the lost copter sound, so it not a scripting issue. Just scripting is the only way you will hit it in the wild 
https://github.com/IamPete1/ardupilot/commit/3cfc6ac72b88e8384e48c447f878361f7f167f8e

WickedShell, 2020-03-23:
This dies because it's a malformed sequence: the `B` sets a repeat, the `L` sets the default note length, but it never plays a note. So it hits the end of the sequence then wants to restart playing the audio again, but since there are no notes it immediately restarts again, which is recursive. This needs to check that any valid note was emitted before allowing a repeat to execute.

IamPete1, 2020-04-01:
This has now been fix in master

closed 2020-04-01 by IamPete1