ArduPilot/ardupilot #13882: EKF2: alt climbing in first arm without takeoff

chobitsfan, 2020-03-25:
## Bug report

**Issue details**
PosHold mode, tested indoor using att_pos_mocap
use RC to arm and then disarm (copter did not take off)
[image]
[log_disarm0.zip](https://github.com/ArduPilot/ardupilot/files/4379249/log_disarm0.zip)
ctun.alt climbing after first armed (throttle min). If you take copter off at this moment, it will rocketing into sky. But this behavior did not happen in following arm

It is strange that it is not happened if log_disarm = 1
[log_disarm1.zip](https://github.com/ArduPilot/ardupilot/files/4379251/log_disarm1.zip)

**Version**
4.0.4-dev

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quadcopter

**Hardware type**
kakute f7

chobitsfan, 2020-03-25:
I am sorry. This seems a duplication of #11304

closed 2020-03-25 by chobitsfan

AGurel, 2020-03-30:
I'm also getting this error. https://discuss.ardupilot.org/t/first-huge-climb-in-althold-flight-mode/54060/2

chobitsfan, 2020-04-21:
Hi @AGurel 
It should be fixed in master by #14030 
Would you be able to test it again? thank you very much