ArduPilot/ardupilot #13889: Continue landing if RTL is triggered during a land

Dan-Moss, 2020-03-25:
###Feature request
Continue landing if RTL is triggered during a land when in auto.

Today we were running auto missions and the battery fail-safe was triggered upon the final decent at about 5m off the ground. The drone elevated to the 15m altitude as per the failsafe, to then land at the same spot but just in RTL rather than land. I know smart rtl may solve this and could be the default in this case?

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2020-03-25:
The `FS_OPTIONS` parameter allows you to specify continue-landing-on-failsafe.  It is not set by default.

Dan-Moss, 2020-03-25:
Not sure how i missed that. Thank you!

closed 2020-03-25 by Dan-Moss